prusa3d/PrusaSlicer #5765: Wrong volumetric flow range in preview mode

mihai-stanimir, 2021-01-13:
### Version
2.3.0+win64 (official release)

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Original Prusa MK3S

### Behavior
In preview mode, the volumetric flow rate visualizer does not use the correct range. This happens often, but not all the time. It used to work fine in version 2.2.0.

Example: I'm slicing two models.

**First model is fine:**
[image]

**Second model is fine:**
[image]

**Both models, range shoots up to 17mm3/s for no obvious reason.** On close inspection no area in the preview is using those red colors:
[image]

Volumetric flow is limited to 11mm3/s so the legend should not go past this value:
[image]

_Is this a new feature request?_
No.
This used to work in 2.2.0 and now it doesn't work anymore.

bubnikv, 2021-01-13:
We will investigate. It may or may not be important, it may be that just a single extrusion with volumetric flow skewed by numerical error could cause this, but we should be sure and possibly improve.

FidelCapo, 2021-01-13:
Can you please attach 3mf file or STLs of objects from this issue to reproduce it? Thank you.

CNCKitchen, 2021-01-14:
> We will investigate. It may or may not be important, it may be that just a single extrusion with volumetric flow skewed by numerical error could cause this, but we should be sure and possibly improve.

Maybe it might a good idea to either allow the user to adjust the upper and lower limit of the color scale by themselves (imo totally necessary for the speed plot due to travel moves being so fast) or use something like the 99percentile as the maximum value to filter out these maybe single extrusions.

mihai-stanimir, 2021-01-14:
Here's a simpler example 3mf:
[PrusaSlicer wrong volumetric flow example.zip](https://github.com/prusa3d/PrusaSlicer/files/5814256/PrusaSlicer.wrong.volumetric.flow.example.zip)

Default settings except I've changed Max volumetric flow to 5, but **legend goes up to 5.4 instead of 5**. With other models or settings it shoots up much higher.

[image]

Another more extreme 3mf example, same model, different print settings, **legend shoots up to 7.9 instead of 5**.
[PrusaSlicer wrong volumetric flow example 2.zip](https://github.com/prusa3d/PrusaSlicer/files/5814288/PrusaSlicer.wrong.volumetric.flow.example.2.zip)

Range varies a lot when changing print settings or printer profiles.

forkineye, 2021-02-27:
I'm experiencing this as well and it's not just a display issue as the gcode is tripping failsafes in Klipper.  Attached is the 3MF file. It seems to be occurring where the small overhang is.  Oddly, this didn't happen yesterday when I printed two of the models at the same time, just when I print a single copy. I'm printing two now to verify.

GCODE leading up to the error:
```
;TYPE:Overhang perimeter
;WIDTH:4
;HEIGHT:4
M106 S255
G1
G1 X144.762 Y167.415 E13.81251
```

Klipper detecting it:
```
Send: N24891 G1 X144.762 Y167.415 E13.81251*91
Recv: // Move exceeds maximum extrusion (11.311mm^2 vs 0.640mm^2)
Recv: // See the 'max_extrude_cross_section' config option for details
Recv: !! Move exceeds maximum extrusion (11.311mm^2 vs 0.640mm^2)
Changing monitoring state from "Printing" to "Cancelling"
```
[18650_1x2.3mf.gz](https://github.com/prusa3d/PrusaSlicer/files/6054871/18650_1x2.3mf.gz)
[image]

neophyl, 2021-02-27:
With mihai-stanimir second example if you turn off gapfill the volumetric rate graph changes to exactly 5mm/2 for the scale.  So its probably related to gapfill.  Is that considered when calculating ?

Forkineye, yours is not just volumetric, the layer height of your red sections jumps from 0.2mm to a whopping 4mm high layer height.  If you scan through the preview you will see those features appear all of a sudden.  So yes printing a 4mm thick layer is exceeding the volumetric rate too.  
This is caused because you have your bridging flow set to 100 rather than the normal value of 1.  I think its doing a nozzle x100 flow, ie 0.4x100 =4mm layer.

forkineye, 2021-02-27:
> Forkineye, yours is not just volumetric, the layer height of your red sections jumps from 0.2mm to a whopping 4mm high layer height. If you scan through the preview you will see those features [...]

That seems to have been my issue, thanks!  It must have gotten set that way from when I pulled my config in from SuperSlicer.  Appreciate you finding it.

lukasmatena, 2021-12-06:
@mihai-stanimir This issue is apparently caused by gap fill, which sometimes generates very short extrusion segments and because G-Code is exported to certain number of decimal places, the resulting volumetric flow in Preview can end up larger than the originally calculated one. This at least is the cause of what is described in https://github.com/prusa3d/PrusaSlicer/issues/5765#issuecomment-760156923. It should be pretty much harmless, only negligible extrusions would suffer from this. We will try to fix it somehow, but not in 2.4 release.

@forkineye Your issue is something else (and simpler). You have set bridge flow ratio to 100, while you probably meant 1. I don't see how you managed to get it there, there is a check that the value is not higher than 2. I saw that your 3MF is from PS 2.3.0, and that check was already in place at the time. Did you do something special to that 3MF?

forkineye, 2021-12-06:
@lukasmatena As mentioned [here](https://github.com/prusa3d/PrusaSlicer/issues/5765#issuecomment-787116099), I think it was due to my config being pulled in from SuperSlicer.

mihai-stanimir, 2021-12-07:
@lukasmatena thanks, main issue is that we sometimes can't tell the printing speed or flow by using the legend and color preview (which can differ from the settings because of the layer time). On occasion the calculated max value is significantly larger than the set max flow, not just by a little bit.

lalten, 2022-09-03:
I've been hitting this issue a lot recently in the form of the `Move exceeds maximum extrusion` Klipper error that cancels the print at a random point.

It turns out this can be completely fixed (in SuperSlicer) by increasing the numerical  numericalGcode numerical precision ("xyz decimals" / "Extruder decimals"):
[image]

This will of course increase the size of the generated gcode and I could imagine it may cause problems for really slow 8bit MCUs. But Klipper should certainly be able to handle it.

It will also fix the pesky autoscaling issue:
* With Gcode precision 3 ("default") the colors are almost meaningless, see the posts above for even worse situations.
  [image]

* With Gcode precision 10 it scales right to 20mm³/s, which is the Autospeed max flow I have set up.
  [image]

MrBlackmidi, 2022-12-15:
Happens to me as well (set limit to 14mm^3)
[image]

bubnikv, 2023-08-08:
PrusaSlicer 2.5.0 and newer improves extrusion rate accuracy by first quantizing the XYZ coordinates and then to round the E axis increment with regard to the segment length which was already quantized to the G-code resolution. While it improves the situation and the volumetric extrusion rate looks much more sane, it does NOT guarantee that the max volumetric rate will NOT be tripped over.

The maximum extrusion width in such case where the E axis delta is rounded up may be clamped either by clamping the E axis delta or the extrusion speed.

eduncan911, 2024-06-16:
For what @lalten mentioned above about SS, it does exist in PrusaSlicer - it's just under Print Settings (not Printer like in SS).

[image]

closed 2026-09-29 by prusaslicer-bot[bot]