prusa3d/PrusaSlicer #5766: Cannot upload gcode to Duet using Windows (works in Linux)

crpalmer, 2021-01-13:
### Version
2.3.0

### Operating system type + version
Windows 10 64-bit

### 3D printer brand / version + firmware version (if known)
Multiple machines using Duet boards.

### Behavior
I am unable to upload gcode to my Duet machines but only from a Windows machine.

To reproduce: open PrusaSlicer 2.3.0, select a physical printer that is configured for Duet.  Click the gear icon to access the physical printer configuration.  Click "test".  I get different results when using the identical configuration on Windows vs. Linux:

Linux: Connection to Duet works correctly.
Windows: Bad Access (error 7)

My configuration is available if needed: https://github.com/crpalmer/3d-printing/tree/master/slic3r

crpalmer, 2021-01-13:
Using WireShark I figured out that it wasn't sending any network data.

Based on that I figured out that Norton was blocking the network access.

I guess you could consider this a feature request to sign your windows executables, for the love of god!

closed 2021-01-13 by crpalmer

bubnikv, 2021-01-14:
> I guess you could consider this a feature request to sign your windows executables, for the love of god!

For the love of God we sign the installers available at https://www.prusa3d.com/prusaslicer/