prusa3d/PrusaSlicer #5767: Suggestion: Make an option to interleave layers to strengthen layer adhesion

Mirarkitty, 2021-01-13:
Idea: Interleave layers so that they bond in z-plane and not just z-axis.

[image]

(Will fix better image later.)

If a perimeter looks like this, it will have a strong bond in the z plane between the layers. This could be a nice idea for a strong 3d print.

Ie, three perimeter shell:

First layer is perimeter 2, 0.1mm high at z 0.1mm.
Second layer is perimeter 1 and 3, 0.2mm high at z 0.2mm.
Third layer is perimeter 2, 0.2mm high at z 0.3mm.
Fouth layer is perimeter 1 and 3, 0.2mm high at z 0.4 mm.
etc

neophyl, 2021-01-13:
Isn't this basically the same as https://github.com/prusa3d/PrusaSlicer/issues/1823 ?

Mirarkitty, 2021-01-13:
Oh yes, I have failed in checking enough. Close this. :D

closed 2021-01-13 by rtyr

Harvie, 2024-02-23:
turn the pattern by 90 degrees (half width on sides, instead of half height at bottom). i think it might seem counterintuitive, but stronger. because it would allow plastic to be squeezed harder in those cracks...