prusa3d/PrusaSlicer #5771: Hilbert Curve Infill: Unable to slice, PrusaSlicer not responding.

jisilva, 2021-01-14:
### Version 2.3.0+

### Operating system type + version
macOS High Sierra Version 10.13.6 (17G14042)

### 3D printer brand / version + firmware version (if known)
Prusa MK3S , Firmware 3.9.3

### Behavior
Unable to complete slicing process when choosing "Hilbert Curve" infill type (top or bottom)
Steps: load STL, choose any printing profile, change top or bottom infill to hilbert curve, slice.
 
Expected Results: slicing complete after a few seconds.
Actual Results: slicing process gets stuck at ~25%, when attempting to cancel, PrusaSlicer stops responding.

_Is this a new feature request?_ NO

[fail.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5811772/fail.3mf.zip)
[image]

bubnikv, 2021-01-14:
It finishes, however it takes too long.

The issue is caused by the improved path planning algorithm, which becomes extremely slow for the Hibert curve to infill. We likely need to limit the expensive "2-exchanges" improvement algorithm to only simpler infill paths or to limit it by time.

[image]

bubnikv, 2021-02-08:
Fixed with 672ed99320ba7e57fa5c1daad3b450ca6320b2f1

Fixed by hard limiting the number of iterations of
reorder_by_two_exchanges_with_segment_flipping()

closed 2021-02-08 by bubnikv

Alucard-i, 2021-04-21:
> It finishes, however it takes too long.
>
>
>
> The issue is caused by the improved path planning algorithm, which becomes extremely slow for the Hibert curve to infill. We likely need to limit the [...]

Have you tried it on a bigger slice? I can do a 200x200 no problem. It takes like 30 seconds but a 320x320 slice can sit for 3 hours and still not be done. The model I have an issue with so not have as much surface as a 200x200 cube but still it won't slice in 3.2.

bubnikv, 2021-07-02:
Fixed with PrusaSlicer 2.3.2-beta.

Alucard-i, 2021-07-02:
> Fixed with PrusaSlicer 2.3.2-beta.

Thank you but sadly I had to move on to other slicer cause this made me not able to make my models I sell. Unsure I will use Prusa slicer again since I am all setup in the new slicer.

nathan118118, 2021-08-19:
I'm trying 2.3.3 and this is still an issue.