prusa3d/PrusaSlicer #5772: --scale-to-fit not accepting any input as valid

mattswest, 2021-01-14:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

[image]

bubnikv, 2021-01-14:
This is a regression likely since PrusaSlicer 2.0.0 when porting the command line interface from Perl to C++.

I have fixed half of the problem for now with 48877614e2138001eda6ff598bbd2b390022d4bf
    Fix of  --scale-to-fit not accepting any input as valid (#5772)
    Fixed parsing of Point3, fortunately Point3 was used at just a single
    command line parameter, thus this was not a big deal.
    
Still the method
```void ModelObject::scale_to_fit(const Vec3d &size) ```
is empty since PrusaSlicer 2.0.0 and it needs to be filled in.

lukasmatena, 2021-08-04:
This issue should be fixed on master by bc81c22. It turns out that not only the `--scale-to-fit` did not work, but the command line interface did not make sure that the objects are on bed and allowed them to sink when transformed. This was probably broken long ago during the Perl to C++ transition.

I reinstated the old behaviour (ensure objects are on bed) and added an extra option `--dont-ensure-on-bed` in case it made someone unhappy. We have just allowed sinking objects in the GUI version, so it makes sense to have it available through the CLI as well. Please, report any problems.

Closing.

closed 2021-08-04 by lukasmatena