prusa3d/PrusaSlicer #5773: Feature Request Layer slider on 3d view

JamesSF69, 2021-01-14:
### Version
2.3.0

### Operating system type + version
Wibdows 10

### 3D printer brand / version + firmware version (if known)
NA

### Behavior
NA

_Is this a new feature request?_
YES
#### Project File (.3MF) where problem occurs
NA

Please let us have a layer slider/visibility layer (like Chicubox) in the 3D view window. This is a great way to see widowed overhangs and be able to paint supports more accurately.

rtyr, 2021-01-14:
Do you mean this?

[image]

JamesSF69, 2021-01-14:
@rtyr Basically.
Not the red part (kinda have those already), but the black lines. And even if no lines, it's nice to drill down the layers to see where islands are.

JamesSF69, 2021-01-14:
[image]

bttguy, 2021-02-05:
+1 Prusaslicer is also a great if not the best SLA slice as well. Hope it will be updated even more!

closed 2026-09-29 by prusaslicer-bot[bot]