prusa3d/PrusaSlicer #5774: Slicing with holes

merc-slk, 2021-01-14:
### Version
2.3.0+win64

### Operating system type + version
Win 10 Pro 1909 build 18363.1316 64bit

### 3D printer brand / version + firmware version (if known)
MK3S + MMU2S 3.9.1+3518

### Behavior
* Import STL (BodemPlaat)
* Add part (Orbis)
* Slice
* Layer 4.20 - 5.00 Holes where it should be 1 compleet layer
* Layer 5.20 change tool

No new request

[Coin.zip](https://github.com/prusa3d/PrusaSlicer/files/5816002/Coin.zip)
[image]
[image]
[image]

foreachthing, 2021-01-14:
You've got ZERO bottom layers. Have at least one and your issue is gone - for me at least.

rtyr, 2021-01-15:
As foreachthing said.

I don't see any issue here.

merc-slk, 2021-01-17:
Tried 1 bottom or 5 bottom layers. Doesn't change anything. The design is on top of the blue circle. The holes shouldn't be there at all.

foreachthing, 2021-01-18:
These "holes" are normal.
Delete the "Orbis" and the holes are gone (for the bottom plate). These holes are by design to save time and material. I'd be complaining if those holes weren't there ;-)
You can't see them after printing anyways...

closed 2026-09-29 by prusaslicer-bot[bot]