prusa3d/PrusaSlicer #5776: Painted Support on internal portion of part not being printed

jsilva-iii, 2021-01-14:
### Version
Version 2.3.0+win64

### Operating system type + version
WIN10 

### 3D printer brand / version + firmware version (if known)
Prusa Mini FW 4.0.3-RC1

### Behavior
Painted support on internal portion of print not working -

[image]

Picture shows the painted support and what was generated, 

[image]

Same part with support on build plate, correct support get's generated
[image]

rtyr, 2021-01-14:
Please provide your 3mf project file.

jsilva-iii, 2021-01-14:
Hi, sorry, I thought it was attached
[WF-Trunk.zip](https://github.com/prusa3d/PrusaSlicer/files/5817497/WF-Trunk.zip)

neophyl, 2021-01-15:
Strange when I open your project and slice, doing nothing else it produces support exactly as expected.  Same version and on the same OS too.

neophyl, 2021-01-15:
It's a lovely bug in the support planner.  I'm not really a coder beyond dabbling now and again but I do a lot of testing at work so interesting problems catch my attention.
If you add a small bit of geometry using a modifier part to 'fill in' the open slot on the bottom layer on the slice so that the inner feature at a higher layer that should be supported is fully 'inside' the model then the support is generated as you would expect it to be.  
[image]

bubnikv, 2021-02-23:
Duplicate of #5948
Fixed with 0367e1cf1b55cd7780785448d2c605ba98025fac

bubnikv, 2021-02-23:
[image]

closed 2021-02-23 by bubnikv