prusa3d/PrusaSlicer #5778: Cannot Clear my Plater

joebuschelman, 2021-01-15:
### Version
_Version of PrusaSlicer used goes here_ 2.3.0

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_ Microsoft 10
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_Ender 3 V2 and Ender 5

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_  I cannot clear my plater.  I can add files and delete them but this object is stuck there.  I have logged off and back on, select all and delete all and it will not clear the plater.   

_Is this a new feature request?_
[image]

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

bubnikv, 2021-01-15:
> I cannot clear my plater. I can add files and delete them but this object
is stuck there. I have logged off and back on, select all and delete all
and it will not clear the plater.

How did you get to that state?

rtyr, 2021-01-15:
It looks like the droid is set as a bed model.

[image]

joebuschelman, 2021-01-15:
I added the file and that is it.  It won’t let me delete it.

joebuschelman, 2021-01-15:
Thank you. I will look at it when I get home from work.

rtyr, 2021-01-18:
Is it ok? Can we close the issue?

rtyr, 2021-01-23:
No response. Closing.

closed 2021-01-23 by rtyr

joebuschelman, 2021-01-23:
I am sorry, I forgot to reply.  Yes it working.  Thank you very much.