prusa3d/PrusaSlicer #5779: Feature Request: Better handling of changing nozzle diameters

amarilio, 2021-01-15:
Changing nozzle diameter right now is a tedious experience:

PrusaSlicer side:
1. Go to extruder setting and change diameter + min + max height limits
2. Go to advanced print settings and modify all extrusion width values (8 in total)
3. Go to Layers and perimeters settings and adjust layer height, and potentially perimeters + solid layers.

Printer side (which also needs to be streamlined):
1. Preheat printer
2. Replace nozzle using two tools
3. Go to printer configuration to change the nozzle diameter
4. Perform a first layer calibration

This is an error prone process with no clear documentation.

Suggestions:
1. Changing nozzle diameter should be a dropdown that match all the sizes the printer selection support (potentially with an option for adding more sizes)
2. Min and Max layer height should either auto populate based on nozzle diameter or be represented in %
3. Extrusion width percentages should be based on nozzle diameter and not layer height (why would you even want it to be as a % of layer height?), and by default be set to be % based to allow easier changes of extrusion width
4. Alternative to #3 - print settings presets are tied to nozzle diameter, so that PrusaSlicer ships with presets that match all the nozzle diameters in the dropdown

bubnikv, 2021-01-15:
We could surely improve PrusaSlicer, however normally you should just
switch the Printer profile, which is linked to Print profiles tuned for the
same nozzle diameter as the Printer profile.
I suppose you own a Prusa printer. What nozzle are you having difficulty
with (missing printer and print profiles)?

> Printer side (which also needs to be streamlined):

   1. Preheat printer
   2. Replace nozzle using two tools
   3. Go to printer configuration to change the nozzle diameter
   4. Perform a first layer calibration

Please post a request for firmware improvement to
http://github.com/prusa3d/prusa-firmware

amarilio, 2021-01-15:
I was not aware of the printer profile solution - that one seems to address most of the concerns. 
It looks like a discoverability problem in the UI. The main culprit is the fact that the UI offers an easily discoverable nozzle diameter in the settings, while the currently intended way to change nozzle diameter is to use a different printer profile.

Ideally these would be combined into a single way to change the nozzle diameter. A simpler stopgap solution to address the discoverability problem could work like this:

Trigger: Users enter a new nozzle diameter into the extruder setting (e.g. 0.25mm). 
New behavior: PrusaSlicer pops up a notification asking the user whether they'd like to switch to the 0.25mm profile that already exists for the printer (if exists).

amarilio, 2021-01-18:
Look at the typical user journey. They get their printer, have only a 0.4mm nozzle, so they select it in the configuration wizard, and never see that wizard again (unless they switch computers).

Once they gain confidence, they'd like to experiment with a different sized nozzle, so they go to settings and see a nozzle diameter settings - but that input setting is a trap, you should never really change it unless you really know what you do (and change all the relevant parameters).

That's the UI problem that should be fixed. My suggestion was simple - if you insist that the right way to define the nozzle diameter is by adding a new printer profile, then at least make it more discoverable in the nozzle diameter setting. A simple solution that I suggested above is that if they typed a number that exist in the preconfigured profiles (e.g. 0.25mm), it prompts them that there is an existing profile for that and ask them if they'd like to use it.

karlosss1, 2021-01-29:
In addition to this same issue, two other issues exist.
1. No print setting user profiles are available unless created from scratch. Thats fine since most of the settings for one nozzle diameter will not suit another but makes issue no 2. even more appaent.
2.  If you make printer profiles for a 0.2, 0.4, 0.6 & 0.8 nozzles then create your favourite Print Setting user profiles for each of them, all the Print Setting profiles you created are available for all the Printer profiles you created.

I don't think you should be able to choose a 0.8mm diameter nozzle printer profile & have a user created Print Setting profile available in the dropdown list that may not suit the nozzle size especially when at the time of creating the Print Settings profile, a 0.4mm nozzle Printer profile was selected.

My suggestion - 
Only display the Print Setting profiles in the dropdown list that were saved for that specific Printer Profile.

closed 2026-09-29 by prusaslicer-bot[bot]