prusa3d/PrusaSlicer #5780: Pause print function shows on different layer than set

wlcina, 2021-01-15:
### Version
Final 2.30

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
MK3S 3.9.3

### Behavior
Setting the pause on print (for inserting bearings) shows on different layer in Slicer, see picture, pause was set at 13.80mm, while the gray area is shown at 14.20mm

[image]

bubnikv, 2021-01-15:
We are aware of this issue, it is caused by the pause print G-code being inserted after Z-hop.
We had a prototype fix, which moved the pause print G-code before Z-hop, but we found it suboptimal.
We will fix it in the upcoming minor release, which we shall do relatively soon.

rtyr, 2022-02-22:
Should be fixed, I am unable to reproduce. Closing.

closed 2022-02-22 by rtyr