prusa3d/PrusaSlicer #5782: Custom G-code for color change does not process macro

fe60, 2021-01-15:
### Version
Version: 2.3.0+win64
Build: PrusaSlicer-2.3.0+win64-202101111315

### Behavior
I want to increase the minimum distance between nozzle and heat bed when filament has to be changed during a print [1]. Therefor I have added a macro to the default G-code `M600` (Printer Settings => Custom G-code => Color Change G-code).
`M600 {if layer_height<70}Z70{endif}`

For layers below 70mm I expect to see the following in the .gcode file.
`M600 Z70` [2]

But instead of getting the Z-parameter, I see this in the .gcode file:
`M600 {if layer_height<70}Z70{endif}`

It seems the macro is not processed. Instead the content is transferred to .gcode file as is.
The documentation about macros doesn't mention restrictions regarding the custom code sections the macros can be applied at [3].

Can you please get the macros working for the "Color Change G-code"?

_Is this a new feature request?_
No

[1] https://github.com/prusa3d/Prusa-Firmware/blob/MK3_3.9.3/Firmware/Marlin_main.cpp#L3169
[2] https://marlinfw.org/docs/gcode/M600.html
[3] https://help.prusa3d.com/en/article/macros_1775

stringlapse, 2021-01-21:
Same issue here on the same version and build. I have `{if layer_z < 30}G1 F300 Z30{endif}` in my custom color change gcode and it's going directly to the gcode file without being processed.

delaosa, 2021-01-31:
Same with version 2.3.0 in Linux, Pause g-code is also ignoring variables.

mvadu, 2021-02-28:
Not only macros, it does not do even simple variable expansion. `M104 S[temperature] comes out as is in the final gcode file.

bubnikv, 2021-03-01:
Implemented with 224f3e506c9db3d2de9e2e441ee4443de2bab964

closed 2021-03-01 by bubnikv