prusa3d/PrusaSlicer #5784: přesah ironingu

vapno45, 2021-01-15:
### Version
_Version of PrusaSlicer used goes here_2.3.0

Dobrý den,

je prosím možné do dalších upgradů PrusaSliceru přidat k "ironingu" funkci "přesah"? Tedy nastavení, které ovlivní procentuální přesah vrstvy ironing do perimetrů. Nyní jezdí extruder při ironingu až ke kraji výtisku a místech, kde je okraj plochy "oblouk" tak jsou tahy ironingu viditelné i přes okraj perimetru.

Díky

[image]

bzed, 2021-01-15:
Translation from google translate:

Good day,

is it possible to add the "overlap" function to "ironing" in other PrusaSlicer upgrades? That is, a setting that affects the percentage overlap of the ironing layer to perimeters. Now the extruder travels during ironing to the edge of the print and in places where the edge of the surface is an "arc", so the strokes of ironing are visible even over the edge of the perimeter.

Thanks

dougemon, 2025-11-26:
Hi there,
Just wanted to say I also would like to request this feature and found this request for it. Thanks!

dougemon, 2026-09-03:
This functionality has not been added to PrusaSlicer 3.0.0 alpha. Ironing overlap amount is still needed