prusa3d/PrusaSlicer #5785: Temp Setting Issue Prusa 2.3.0

Andre6553, 2021-01-15:
### Version
_2.3.0_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_ When i set printing temp on my ender 3 pro 230deg C in slicer the gcode generated show it must be 226deg c
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2021-01-15:
Please provide some description of the issue, 3MF project file, etc.

rtyr, 2021-01-28:
No response. Closing.

Likely same as https://github.com/prusa3d/PrusaSlicer/issues/5898.

closed 2021-01-28 by rtyr

Andre6553, 2021-01-29:
Hi i manage to fix it by delete my profile on PS and setup a new print profile. I. Did not save the 3mf file.