prusa3d/PrusaSlicer #5786: Some Layers/Walls Disappearing

MarcAhoy, 2021-01-15:
Prusha Slicer 2.3-Win64

Windows 10 

Anycubic i3 mega-s

Walls of the model are there in the preveiw, but after slicing, in the slicing view, the model is missing some walls/layers

rtyr, 2021-01-15:
Please provide your 3mf project file.

MarcAhoy, 2021-01-19:
[New Compressed (zipped) Folder.zip](https://github.com/prusa3d/PrusaSlicer/files/5839028/New.Compressed.zipped.Folder.zip)

rtyr, 2021-01-19:
The missing walls are too thin to be printed with your extrusion width (0.82). Try 0.68 for perimeters and external perimeters, it should fix your issue (and it is also quite appropriate since you have 0.6mm nozzle). You can also try "Detect thin walls" parameter, but it can introduce different kind of issues.

Closing.

closed 2021-01-19 by rtyr

MarcAhoy, 2021-01-20:
oh, thanks. problem solved, the layers are there now