prusa3d/PrusaSlicer #5790: Bug: Network Error - PrusaSlicer Standalone Installer

tburde, 2021-01-15:
### Version

2.3.0 Standalone Installer

### Operating system type + version
Windows 10 Version 10.0.19042 Build 19042
Processor:	Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 2208 Mhz, 6 Core(s), 12 Logical Processor(s)
Ethernet:		[00000003] Realtek USB GbE Family Controller
WiFi:		[00000004] Intel(R) Dual Band Wireless-AC 8265
iGPU:	Intel(R) UHD Graphics 630
dGPU:	NVIDIA GeForce GTX 1050 Ti with Max-Q Design

### Behavior
When running the standalone installer, there is a "network issue" but everything seems to work for other programs and web-installers. I have tried using WiFi and Ethernet. I was able to install 2.3.0 by using the Apps & Drivers version. I tried multiple downloads across 2 days along with some restarts too.

[image]

rtyr, 2021-01-16:
What antivirus software are you using? Does it happen if you temporarily disable it?

neophyl, 2021-01-16:
Several other people on the forum have had this issue too https://forum.prusaprinters.org/forum/prusaslicer/cannot-install-2-3-0-final-says-i-dont-have-internet/

The consensus there is that the stand alone installer is broken as the normal installer works fine when they try that.

I always run Github versions so I haven't seen it myself.

tburde, 2021-01-16:
I was using Norton Antivirus before but at the moment just the default Windows Defender.
I have no Proxy enabled and I have tried with and without VPNs.
I also tried disabling the Windows Firewall temporarily and it had the same error.

tburde, 2021-01-16:
@neophyl I have seen a few people report that, but it seems to work for some others so I would be curious to know what the issue could be! I think it is because the normal installer downloads all the files within the installer?

rtyr, 2021-01-16:
Normal installer does not require the internet connection, unlike the standalone installer.

howardjones, 2021-02-06:
I am also seeing this with the Standalone installer only, on Windows 10, 19041. The full installer works fine, although it's an additional 300MB of download I didn't need! Internet is working fine.

Janys, 2021-04-19:
Hello, we have disabled the internet connection check in the new 2.3.1 version. Can you please check whether your issue is fixed?

tburde, 2021-04-22:
I am happy to report that the installer for 2.3.1 worked! It installed the latest version and also uninstalled the last version (2.3.0) and kept all my settings/profiles.

closed 2021-04-22 by tburde