prusa3d/PrusaSlicer #5791: Same layer infill and perimeter does not show in one plane in Preview mode

lukasradek, 2021-01-15:
### Version
2.3.0+win64

### Operating system type + version
Win10

<details>
  <summary>System Info</summary>
  `
PrusaSlicer
Version:   2.3.0+win64
Build:     PrusaSlicer-2.3.0+win64-202101111315

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19041), 64-bit edition
Total RAM size [MB]: 17,109MB
OpenGL installation
GL version:   4.6.0 NVIDIA 425.31
Vendor:       NVIDIA Corporation
Renderer:     GeForce GT 750M/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
[... 284 lines cut ...]
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
`
</details>

### Behavior
In preview mode, when examining infill and perimeters, you can notice, that the farther the point of examination is from your viewpoint and the steeper the angle you are observing from, the more the infill and perimeters appear to be in different layers.

Look at the farther perimeter. The infill is at the same layer as the perimeter, yet they are rendered appart at the farther point, while they are rendered on the same layer at the closer point.
[image]

Same place, different angle. The infill is now rendered "at much closer layer" to the perimeter at the farther point.
If I set my line of sight to be parallel with plate, the infill is rendered correctly.
[image]

rtyr, 2021-01-15:
Please provide your 3mf project file.

lukasradek, 2021-01-16:
I am sorry, that was completely failure of my brain.

I looked at it with fresh mind this morning, tried to construct a minimal example and realized that I have switched "Infill only where needed" and so the infill was not actually reaching the perimeter, but since it was somewhat following the perimeter geometry, it fooled me 🙂.

So at least... all the best in new year.

closed 2021-01-16 by lukasradek