prusa3d/PrusaSlicer #5792: PS 2.3.0+ Layers & Perimeters Modifier Behavior Bug

ctilley83, 2021-01-15:
### Version
2.3.0+ final

### Operating system type + version
Mac OSX 11.1

### 3D printer brand / version + firmware version (if known)
RRF 3.2

### Behavior
I have a "Layers and Perimeters" modifier on a part surrounding letters.  This is so the letters have 2 perimeters, but the remainder of the part have 4 perimeters. This works as intended other than the fact that the slicer wants to add 4 perimeters around the modifier. The edge of the modifier should be ignored and the slicer should not attempt to count that as a physical feature. It's a modifier not a part.
[image]
Results in:
[image]

_Is this a new feature request?_
Bug

#### Project File (.3MF) where problem occurs
[Voron_Blank_LCD.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5823318/Voron_Blank_LCD.3mf.zip)

rtyr, 2021-01-16:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/3846, https://github.com/prusa3d/PrusaSlicer/issues/3317 and some other linked issues.

The parameters set for the modifier are not the same, so it splits the object into pieces, which are then sliced independently. That is the reason, why the area covered by the modifier gets its own perimeters. See the related issues for more information. It can be likely improved (like everything), but this is normal behavior at the moment.

bubnikv, 2021-06-03:
Duplicate, closing this one.

closed 2021-06-03 by bubnikv