prusa3d/PrusaSlicer #5794: Impact of supports on cost and print time

thehumanfallacy, 2021-01-16:
### Version
2.3.0 / latest

### Operating system type + version
All

### 3D printer brand / version + firmware version (if known)
Applies to all models.

### Behavior
* _Describe the problem_
  * It would nice to be able to see the cost and printing time of the supports that have been added to the slicing.  The cost should be displayed in the _Slice Info_ section just like Used Filament (m), Used Filament (m), Used Filament (mm3) and Used Filament (g).  The printing time could displayed in a new section so you can see how much time the supports are adding to the print.  This would be nice to see if tweaking support optimizes the printing time and filament consumption.
* _Actual Results_
  * No way to see cost or additional print time added by supports.
* _Steps needed to reproduce the problem_
  * Perform regular slicing with supports.
* _Expected Results_
  * New metrics are displayed.
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_ Yes

#### Project File (.3MF) where problem occurs
N/A

rcmaniac25, 2021-01-16:
You can see the time spent on supports in the preview window: https://blog.prusaprinters.org/prusaslicer-2-3-is-out-paint-on-supports-ironing-and-more_42475/#print-time

thehumanfallacy, 2021-01-16:
Ah yes, you're right, that's handy!  Maybe we could also add the amount of extruded filament to this table.

rcmaniac25, 2021-01-16:
Right now it seems like it's just times, percentage of total print time, and a toggle to hide/show in the preview. Not sure about moving other stats to it. (also, even if you do multi-extruder, the existing filament usage shows per-extruder)

bubnikv, 2021-10-05:
Implemented don't know when
[image]

closed 2021-10-05 by bubnikv