prusa3d/PrusaSlicer #5795: Toggle for showing only physical printers

freiser77, 2021-01-16:
### Version
2.3.0

### Operating system type + version
macOS Big Sur

### 3D printer brand / version + firmware version (if known)
MK3S MMU2S

### Behavior
I only have one 3D-Printer and would like to be able to restrict the printers selectable in the Plater view to only the physical printers. In my case this would be the Single Printer in 0.4 and 0.6mm nozzle, as well as the MMU version in 0.4 and 0.6mm version.
Yes, this is a new feature request.

fe60, 2021-01-16:
Isn't the last entry in the list of printers what you are looking for?
[image]

Alternatively you can start the Configuration Wizard via the menu.
Configuration => Configuration Wizard

freiser77, 2021-01-16:
Hello @fe60,

I want to be able to hide all those presets in exactly this dropdown.
So only the physical printers (created with your highlighted item) are shown.

Greetings

fe60, 2021-01-16:
That's what the configuration wizard does for you. Selecting the last entry in the dop-down leads to an option to add or remove presets. And choosing this option will open the wizard. Just deselect the models/nozzles you want to disappear and finish the wizard.

freiser77, 2021-01-16:
Hello @fe60,

I don't want to delete the models from disk. I simply don't want to show them in the dropdown.

fe60, 2021-01-16:
That's what the wizard does. When you want to get some entries back, just mark the printer nozzle combination(s) you are interested in and you will see them again in the drop down.

Nailig, 2023-10-28:
@fe60 if you make your own physical printers they depend on the system default printer for settings that you did not change. If there are updates on the default printer they are enclosed in the user created profile (if the user did not edit that parameter)

I would like the same thing: hide certain system presets (because I tweaked them) from the printer selection in the main window. and keep them in the printer settings window

closed 2026-09-29 by prusaslicer-bot[bot]