prusa3d/PrusaSlicer #5796: Upgrade to 2.3 deleted my print settings, strange color change at first layer

lindbergalec, 2021-01-16:
### Version
2.3

### Operating system type + version
MacOS Catalina 10.15.7

### 3D printer brand / version + firmware version (if known)
Creality Ender 3 Pro

### Behavior
* Upgraded to 2.3, and it erased my print settings, left it with a default that it then gave an error for saying that the first layer height cannot be more than the nozzle diameter.
* Also I'm getting a color change at the first layer, as in printer will print the priming line then stop and retract for a color swap when I had not asked for one
* Naming Schema seems to have changed when saving - from "50mm_xyz_cube_0.2mm_PLA_ENDER3_56m" to "50mm_xyz_cube". I have not found a way to change back.

* _Steps needed to reproduce the problem_
  * N/A
* _Expected Results
* Don't delete user data
* Should not have a color change when one was not requested and I cannot figure out how to remove it so I'm going to have to use a different slicer, especially since my print settings were wiped.
* _Actual Results_
* Described in behavior
* Restarting & going through the setup again seems to have resolved the 1st layer color switch after resolving the .3 and .35mm first layer height issue. But the removal of my presets was not resolved.

_Is this a new feature request?_
No

lukasmatena, 2021-01-16:
What was the previous version of PrusaSlicer that you were using? Was it any of the 2.3 alphas/betas?

lindbergalec, 2021-01-16:
Hi,

I believe so, looking at my downloads, I see a 2.3 version from April

zapdogster, 2021-01-16:
I was using 2.3 RC3 on a Win 10 x64 machine. After installing PrusaSlicer-2.3.0+win64-202101111315 all of my print setting are gone except for two random ones I never use. I cannot seem to even restore all of the system ones, and there are none listed under print settings.
[image]

zapdogster, 2021-01-16:
Update - I got all of my previous print settings back by activating a previous Configuration Snapshot.

lukasmatena, 2021-01-18:
@lindbergalec The first alpha of 2.3.0 was released in October. Anyway, the reason I asked is that alphas and betas use different config directory than the rcs and finals, so I wanted to rule out a possibility that you created your profiles in a beta and are now surprised that they are gone. That would be normal (profiles need to be exported/imported in that case). But @zapdogster 's reply shows there is indeed something wrong.

@zapdogster Thanks. Could you please zip and send us your complete configuration directory (unless it contains some data you don't want to share) so we can have a look at what might have happened? (Help->Show Configuration Folder).

@lindbergalec Can you also get your profiles back by reverting to previous snapshot (Configuration -> Configuration Snapshots)? If so, could you please also send the zipped directory?

zapdogster, 2021-01-18:
HI - So turns out restoring the Configuration Snapshot was only a temporary fix. The Print Setting disappeared the next time I started up the Slicer. I got them back permanently by:
 - Import Config from Project (File->Import)
 - Saved them for "next time" using Export Config Bundle

However, sometime they seem to disappear again - not entirely sure on that as sometimes it comes back and other times it disappears??? So I'm not entirely sure this will work 100%. Tracking this behaviour.

zapdogster, 2021-01-18:
[PrusaSlicer.zip](https://github.com/prusa3d/PrusaSlicer/files/5830594/PrusaSlicer.zip)

Here are my config files after I restored the Print Settings.

rtyr, 2021-01-18:
@zapdogster I can see that most of your custom profiles have 0.4mm nozzle diameter condition, so such profiles will not be visible for your custom "0.5mm Nozzle X" printer profile.

Only those two profiles you mentioned earlier are compatible with 0.5mm nozzle.

[image]

[image]

zapdogster, 2021-01-18:
Ahh - that was unknown to me. 

So in mine, when I unclick All in the Dependencies, the "Compatible" line gets grey, and that print setting reappears in the Print Settings list for the 0.5mm nozzle. Personally I find the interaction of the two settings a bit confusing at this point, but I can work around that. Thanks again!!!

[image]

I am guessing I could just remove the printer condition for the Print Settings I would like to be generic to all Printer Settings?

Also, **what does the Detach From System Preset do?** and why would I use it? I read some post in the forums but don't quite understand it. That setting appears in at least 3 places: Print Settings. Filament Settings, and Printer Settings.
[image]

rtyr, 2021-01-18:
You can just remove  `and nozzle_diameter[0]==0.4` from the condition and save the profile. But since you can't use that profile anyway (incorrect settings for 0.5mm nozzle), you should rather make a separate profiles for 0.5mm nozzle with  `and nozzle_diameter[0]==0.5`

[image]

zapdogster, 2021-01-18:
Hi - so I think you are saying that in the Print Profiles, I should make new "copies" with parameters for a 0.5 nozzle and change the Compatible Printer Conditions for 0.5mm?

I did this along the way, and it seemed to work:
[image]

I'm pretty new at this, but there seem to be a large number of "Dependency" options and it is very unclear to me how they interact (or don't interact)...

Thanks
gny

rtyr, 2021-01-18:
Don't do anything with "Compatible printers", just edit the condition.

rtyr, 2021-01-19:
> 
> Hi, I believe so, looking at my downloads, I see a 2.3 version from April
> […](#)

@lindbergalec What was the filename?

lindbergalec, 2021-01-22:
Reverting did seem to return the profiles, find attached, restarting the
slicer also seemed to keep the returned settings.
[image: Screen Shot 2021-01-21 at 4.14.48 PM.png]

rtyr, 2021-01-28:
I guess we can close this now.

closed 2021-01-28 by rtyr