prusa3d/PrusaSlicer #5797: numeric keypad behavior

duebelpapst, 2021-01-16:
### Version
Version: 2.3.0+win64
Build: PrusaSlicer-2.3.0+win64-202101111315
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19042), 64-bit edition
Total RAM size [MB]: 17,092MB
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
Prusa MK3S MMU2S / 3.9.3
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
When I enter something in a text field with the numeric keypad, for example with the height range modifier, often the number does not appear in the text field, but the view of the printing plate changes according to the key, with 1 top view, with 2 bottom view, and so on.
It makes no difference whether I double-click or tab into the field, only after several attempts does it work.

* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

fe60, 2021-01-16:
It's not related to the Numpad. The focus of the input fields in the sidebar gets lost, as soon as the mouse pointer is moved back to the Preview.

How to reproduce:
0 - put an arbitrary object on the plater
1 - select it
2 - set focus on one of the input fields defining the location or size of the object
3 - move mouse pointer back to the preview
=> Input field lost focus. Pressing any number will control the viewing angle of the preview instead of changing the value in the input field.

[image]

fe60, 2021-01-28:
@duebelpapst Can you please clean up your initial post?
This will make it easier for the Prusa staff to process it and I would like to see this getting fixed.

Some proposals:
Change **Title** to describe the "real" issue
Focus of input fields in right panel gets lost when mouse pointer is moved back to preview

**Version**
Remove the explanations for issue creation (italic)

**Operating system**
Remove the explanations for issue creation (italic)

**3D printer brand**
The whole section can be removed, as it doesn't matter for the issue.

**Behavior**
Remove the explanations for issue creation (italic)

**Project File (.3MF)..**.
The whole section can be removed, as it doesn't matter for the issue.

rtyr, 2021-03-02:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2519. Closing.