prusa3d/PrusaSlicer #5798: Issues with .3MF-files

Cub3r4, 2021-01-16:
### Version
2.3.0 stable

`About->About PrusaSlicer`

### Operating system type + version
Windows 10 Pro 1909

### 3D printer brand / version + firmware version (if known)
Original Prusa MK3S and others (doesn't matter for this request)

### Behavior
I export my meshes with Solidworks 2020.
When I import .3MF-files in PrusaSlicer, PS does not matter about the filename of the .3MF.
Instead, it takes the name given in the model-file in the .3MF (3MF's are ZIP's).
There is several metadata and also a name="..." in the .model-file in the .3MF.
So for example to describe what I mean I show you the difference of importing a STL and 3MF in my case:

Examplefile.STL     ->     shows up in PS as "Examplefile.STL" in the list in the right
->     when exporting default name is "Examplefile.gcode" (with no additional output filename format)
->     I save the file

Examplefile.STL     ->     shows up in PS as "body32224" in the list in the right
->     when exporting default name is "body32224.gcode" (numbers differ from file to file)
->     I have to rename it to "Examplefile.gcode"
->     I save it

I know that this in first place a issue with Solidworks (there are no options in SW to change the behaviour with the 3MF metadata)
but I think there should at least be a checkbox in the configuration settings of PS where you can choose what name is importet with .3MF files because I think most people just want to have their filename shown up in PS.
I like that .3MF are able to store various metadata but usually you name exportet meshes useful so that you can work with them in your file explorer and understand what file is what (simple). And not care about a name in the metadata.
It is a little annoying to always rename the .gcode after slicing if you have already done that when you saved your mesh.
It really disturbs the workflow...

A second issue I am having with .3MF's is the placement of the object on the buildplate after importing in PS.
STL's are always dropped in the middle of the buildplate.
3MF's are placed at 0, 0, 0.
It is not as annoying as the first problem with the names but also related to the metadata of the 3MF-format.
This placement issue also occurs with Cura, the naming issue however not.

**In conclusion I would be happy seeing a checkbox in the configuration settings, that lets the 3MF's be treated like STL's and ignores the metadata of the 3MF's (naming and (if possible) object placement).**

_Is this a new feature request?_
Yes :-)
[image]

Here a .3MF exported with SW2020:
[Example 3MF.zip](https://github.com/prusa3d/PrusaSlicer/files/5824599/Example.3MF.zip)

rtyr, 2021-01-16:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/4229.

calebreister, 2021-05-22:
This issue seems to hold true for *all* 3MF files. It doesn't matter if you imported multiple STLs and saved them in a 3MF, or opened an existing 3MF file; PrusaSlicer always uses the name of the first item in the tree when you first try to save. While this is a sensible default when you are first creating/saving a 3MF file or exporting from a file that has not yet been saved, it doesn't make sense if the 3MF file was opened from a file manager, edited, then saved.

devsfan1830, 2021-08-12:
I'd just like to add in my vote to have this fixed. Fusion360 now defaults to 3MF for export and I'd like to keep using that. But in my case when importing a 3MF, every single file exported as such by Fusion360 a show up as "MeshBody1" under the model list on the right. This seems to correspond to the name given the "body" element within the fusion360 project rather than the file name I gave the exported file. For those not in the habit of labeling every single "body", PS should be using the file name.

bubnikv, 2021-10-04:
Would you please verify that you are happy with the current behavior implemented in PrusaSlicer 2.4.0-alpha3?

Cub3r4, 2021-10-04:
**I can confirm the naming issue as fixed (PrusaSlicer 2.4.0-alpha3).** I tried 3MF-files containing one (e.g. the file from the issue-report) and also containing multiple bodies. Now when exporting, the filename is chosen. Nice! Thank you very much!

To the placement issue:
The 3MF-file is now successfully placed in the middle of the buildplate, like a STL.

Most of my tested 3MF's are dropped onto the buildplate. However thats not always the case. Here some examples:
[Flex Back.zip](https://github.com/prusa3d/PrusaSlicer/files/7277227/Flex.Back.zip)
[Backplate.zip](https://github.com/prusa3d/PrusaSlicer/files/7277426/Backplate.zip)
But just one click on the button "drop to bed" in the right and its there just like the STL.

For further interest I can add the information, that the origin of coordinates in SW2021 seems to have nothing to do with that.

On one hand, I'd like to see this last point also be fixed since both filetypes should be treaten the same ("in theory") when importing them into PS and containing the same model.
But on the other hand I don't really mind. Because in my case (SW2021) y- and z-axis are exchanged and I always have to click on "place to face" to rotate it. And then it makes no difference in what height the model was before.

**So if noone minds, I will consider my report as closed.**

closed 2021-10-05 by Cub3r4