prusa3d/PrusaSlicer #5802: unable to install slicer in OSX 10.10.4

out2sea71, 2021-01-16:
### Version
Prusaslicer 2.3.0

Mac OSX 10.10.4
Unable to install the dmg file.  It shows the screen to "drag to applications file" but it doesn't do anything else.  I can find no other instructions or tips on further actions.  Please help.  Thank you.

I use Prusa Mk3s+ printer and the slicer I use with windows does just fine.  I have just recently this morning updated the slicer and printer to latest the internet is pushing out..

Thanks again for anything guidance int he solution to the problem.

out2sea71

out2sea71, 2021-01-18:
> Just drag "Original Prusa Drivers" printer icon on "Applications" folder icon. It will copy all files to your Applications folder. Then you can run slicer from this folder or from "Launchpad".

Thank you for the response but this does nothing.  It doesn’t show up in launch pad.

FidelCapo, 2021-01-19:
When you double click on the "Applications" folder from my screenshot, will it open your "Applications" folder?
Do you have an "Original Prusa Drivers" folder in your "Applications" folder?

out2sea71, 2021-01-23:
yes, that is there

branog, 2022-06-21:
I had the same issue. I finally tried to DELETE the whole folder "Original Prusa Drivers" from Application folder and tried again "DRAG TO INSTAL " and it works. All profiles, settings have been preserved! Good luck ;)

closed 2026-09-29 by prusaslicer-bot[bot]