prusa3d/PrusaSlicer #5803: Solid infill should use selected fill pattern

jbro-oregon, 2021-01-16:
### Version
2.3.0+win64

### Operating system type + version
Windows 7 Professional 64-bit

### 3D printer brand / version + firmware version (if known)
Prusa mk2s FW: 3.2.3

### Behavior
New feature request:
Solid infill above bridged non-solid infill is rectilinear instead of selected fill pattern. It seems to me that solid infill should match the fill pattern selected, or at least have its own fill pattern selection pull-down instead of being stuck as only rectilinear with settable angle.

#### Project File (.3MF) where problem occurs
This is universal, not specific to a file. Slice anything and look at results to witness what I am describing.

rtyr, 2021-01-18:
Why do you need it?

Most of the fill patterns are not suitable for 100% density.

andreyoleynik, 2021-01-19:
Sometimes I need circular pattern or something like this

jbro-oregon, 2021-01-19:
In some cases, I need concentric infill for best results. Having the solid layers stuck at rectilinear takes longer to print and reduces print quality. An example is a long and thin rectangle; the rectilinear back-and-forth path is very short, which can cause extrusion defects and takes longer to print than infill aligned with the long axis of the rectangle (like concentric).

SeeJayDee, 2021-03-10:
I second this as a feature request. Being able to set the solid infill to one of the existing top- or bottom-infill patterns would allow better control over print defects and could also increase print speed. Use cases like @jbro-oregon's example are not uncommon.

carlosjln, 2021-06-06:
> Why do you need it?

@rtyr Because I don't want the printer to zig zag when it could perfectly do concentric lines and be quicker with less issues.

[image]

kuon, 2022-05-19:
This seems similar to #2147 an #6014, maybe we can consolidate in one issue?

This is quite an important feature for me, is there any workaround?

michael-from-there, 2022-06-18:
> This seems similar to #2147 an #6014, maybe we can consolidate in one issue?
> 
> This is quite an important feature for me, is there any workaround?

      Try this

Infill: 0%

Vertical shells:
   Perimeters:  ABIGNUMBER

carlosjln, 2026-09-14:
This feature is still needed!

wikolii, 2026-09-29:
Still needed

Jan-Soustruznik, 2026-10-06:
duplicate of #2147

closed 2026-10-06 by Jan-Soustruznik

guiweber, 2026-10-09:
FYI - The issue this was a duplicate of (https://github.com/prusa3d/PrusaSlicer/issues/2147) has been closed and moved to a discussion here: https://github.com/prusa3d/PrusaSlicer/discussions/15999

If this feature is important to you, go to the discussion to upvote it and provide examples of where it can be useful.