prusa3d/PrusaSlicer #5806: Avoid crossing perimeters crosses holes

jbro-oregon, 2021-01-16:
Version
2.3.0+win64

Operating system type + version
Windows 7 Professional 64-bit

3D printer brand / version + firmware version (if known)
Prusa mk2s FW: 3.2.3

Behavior
Setting to avoid crossing perimeters still crosses holes within the layer. This leads to stringing in holes where it is hard to clean out post-processing. The slice preview correctly shows hole diameters as "External perimeter".

Project File (.3MF) where problem occurs
This is universal, not specific to a file. Slice anything with a hole and look at preview results with travel shown to witness what I am describing.

jbro-oregon, 2021-01-16:
OK, I do see that in some cases it does what it is supposed to do. So here is a .3mf that causes the problem. All holes are breached, but the tip holes are grossly crossed.
[3mm spacer.zip](https://github.com/prusa3d/PrusaSlicer/files/5825293/3mm.spacer.zip)

rtyr, 2021-01-18:
Reproduced. Thank you for reporting it.

hejllukas, 2021-01-19:
I looked at your 3MF file, and this behavior is caused by setting `Minimum travel after retraction` to 5 mm in `Filament Overrides`. This means that retraction isn't used for travels shorter than 5mm. But the avoid crossing perimeters is used only for travels, which use retraction. In your 3MF file, travels that are crossing holes are shorter than the value of `Minimum travel after retraction, so the avoiding crossing permits is not applied to them.

This is intentional behavior because the avoid crossing perimeter could make travels (even short) much longer than was initially. Also, z-hop is disabled when retraction isn't used. So if the avoid crossing perimeters will be used independently on retraction, it could cause more problems than benefits.

To get rid of these travels, you could try to decrease the value of `Minimum travel after retraction` (in your case in `Filament Overrides`). I tested the value 3.5mm, and this should be enough to get rid of these travels crossing holes in your case.

jbro-oregon, 2021-01-23:
Thank you for this information. It makes sense.
It would be great to have a separate option to just not cross holes. The IdeaMaker slicer for the Raise3D has this option, and it is very useful to get good hole quality and keep overall print times lower than complete avoidance of all perimeters.

sbierly, 2023-09-28:
Hello, 

I was happy to find this thread and apparent solution here at the bottom, from over 2 years ago, as I've been constantly plagued with tedious cleanup when printing TPE parts. In general, I think I have this down, but the strings in holes issue has bee super annoying, particularly now that I'm trying to make a lot of these tiny washers. The above fix does nothing to change the problem. You can see in the attached photo and test plate what happens. The first part always prints fine, but the problem seems to be that the tool then moves to the geometric center of the subsequent parts (breaching perimeters), and then travels to a perimeter to begin printing the part. Perimeters are only breached at the beginning of each part, for the 2nd+ parts. 

I left the attached example set to the default value of "Minimum travel after retraction" in Filament Overrides as 5 (this was a Prusa standard setting for TPE), but changing it even down to zero did nothing but add retractions, the ooze in the holes was still there. Unless you have another setting up your sleeve that's new to me, this seems to be a bug.

[image]

[Hole test plate_0.2mm_FLEX_MINI_20m.zip](https://github.com/prusa3d/PrusaSlicer/files/12752877/Hole.test.plate_0.2mm_FLEX_MINI_20m.zip)

closed 2026-09-29 by prusaslicer-bot[bot]