prusa3d/PrusaSlicer #5811: Slicer closes intentially left open areas  - eg. top or holes

jlumera, 2021-01-17:
### Version
2.3.0+win64

### Operating system type + version
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19041), 64-bit edition
Total RAM size [MB]: 34,297MB

### 3D printer brand / version + firmware version (if known)
Prus MK3s Stock

### Behavior
expected behavior - dont close intentionally left open areas with walls or areas of walls
real behavior - top surface close and certan holes on the bottom are close with walls

see screenshot before slicing and after

a test with Simplify3D did not show those problems ( so i assume the model is OK)
the model was locally generated with SCAD based on this thingiverse model https://www.thingiverse.com/thing:735813

#### Project File (.3MF) where problem occurs
[image]
[image]

Here is the project file
[strainer56x31x15_Miele.zip](https://github.com/prusa3d/PrusaSlicer/files/5826242/strainer56x31x15_Miele.zip)

neophyl, 2021-01-17:
Loaded up your 3mf file into 2.3.0 on win 64 so the same version and OS you are running.  PS reports its fixed 252 issues with the model and slices it correctly.
I suspect the save to 3mf has fixed a few things so you might want to try loading your own saved 3mf file.

When dealing with Openscad files its a fair assumption to make that the model has issues.  Even if another slicer works with it. 
So many of the broken models we see on the forum are from scad.  Prusa Slicer doesn't seem to handle them as well as say Cura or S3D unfortunately.  They are however broken.  Often the right click - fix through netfabb option is enough to make them printable though.

Even after the repair there are still 291 intersecting faces, 193 zero faces, 174 non flat faces and 112 thin faces.  PS seems to cope with those though.

[image]

jlumera, 2021-01-18:
Many thanks - the saving (and loading) of the 3MF really fixed the issue as you described.
Another reason to go completely to 3MF. 
My assumption also was that the model causes the problem - but was unaware how the different the behaviour is between the slicers.

One comment: would have been a good hint if the slicer would have provided a feature to check the model and fix it if necessary.  conformity check ? Similar what happened when the 3MF file was loaded.

closed 2021-01-18 by jlumera

FidelCapo, 2021-02-01:
I can't test the fix for this model because there is no stl file available, only a 3mf file. However, according to tests of other models, this issue should be corrected in next release.

bubnikv, 2021-07-02:
Fixed with PrusaSlicer 2.3.2-beta.