prusa3d/PrusaSlicer #5812: prusaslicer wont open 2.3 windows 10

poblopuablo, 2021-01-17:
### 2.3

### windows 10 pro build 19041.746
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### prusa mk2.5s

### Behavior
*Describe the problem: _when i try to open PS it pops up this error every time. i had a system crash earlier and had to shut the pc down now this pops up_
* Steps needed to reproduce the problem: _attempt to click on "prusa-slicer.exe"_
  
* Expected Results: _open prusa slicer_
* Actual Results: _PrusaSlicer GUI initialization failed pop up window_
* 
[image]

*   refrence: (Won't Open - Windows 10 #4915 had a similar issue but that was on the alpha im on full release i believe)

rtyr, 2021-01-18:
Your PrusaSlicer.ini file is corrupted, it is likely caused by the system crash you mentioned. Delete the PrusaSlicer.ini file and restart the PrusaSlicer.

poblopuablo, 2021-01-18:
> Your PrusaSlicer.ini file is corrupted, it is likely caused by the system crash you mentioned. Delete the PrusaSlicer.ini file and restart the PrusaSlicer.

where can i find that file?

bubnikv, 2021-01-18:
the path is in the screenshot you posted.

poblopuablo, 2021-01-18:
thanks guys! i did end up loosing some of my settings but its back! =]

closed 2021-01-18 by poblopuablo

bubnikv, 2021-07-02:
Transactional saving of PrusaSlicer.ini implemented in PrusaSlicer 2.3.2-beta.