prusa3d/PrusaSlicer #5813: Remove unnecessary null pointer checks

elfring, 2021-01-17:
[Extra null pointer checks are not needed in functions](https://isocpp.org/wiki/faq/freestore-mgmt#delete-handles-null "Do I need to check for null before delete p?") like the following.
- [BitmapTextRenderer](https://github.com/prusa3d/PrusaSlicer/blob/7f22ce63f65e2af48631eaed0ec1c396a373ea76/src/slic3r/GUI/ExtraRenderers.cpp#L48 "Destructor for BitmapTextRenderer")
- [Sidebar::priv](https://github.com/prusa3d/PrusaSlicer/blob/5b138cf04d13e05e2f8f5dab5bcb2375fbf289ce/src/slic3r/GUI/Plater.cpp#L596 "Destructor for Sidebar::priv")
- [View3D](https://github.com/prusa3d/PrusaSlicer/blob/7d751753bc22e3d0c0355e2a940c15db6215ef08/src/slic3r/GUI/GUI_Preview.cpp#L37 "Destructor for View3D")

bubnikv, 2021-01-18:
You are right. I have fixed the spots that you referred to.
It is harmless, but could be better.
A pull request would be better.

closed 2021-01-18 by bubnikv

elfring, 2021-01-18:
Thanks for [your small source code improvement](https://github.com/prusa3d/PrusaSlicer/commit/9a2310ae98d78763b92c8f25c893e2e721b7de91 "Deletion of a few redundant checks").

How do you think about to [use a development tool like “clang-tidy” for further adjustments](https://clang.llvm.org/extra/clang-tidy/checks/readability-delete-null-pointer.html "Deletion of unnecessary null pointer checks")?

bubnikv, 2021-01-19:
Thanks for the hint. There are so many things we would like to do, but the
time is the constraint.
In the first place, we should get rid of compiler warnings.

lukasmatena, 2021-01-19:
> we should get rid of compiler warnings

Which I'm already working on. (Just so anyone keen on making a PR knows.)