prusa3d/PrusaSlicer #5814: PrusaSlicer 2.3.0 rc2

gcarissimo, 2021-01-17:
[image]
[image]

2.3.0 rc2

Versión 2.3.0-rc2+win64
Windows 10

I get an error as soon as the PrusaSlicer starts and the mouse moves.
Can anybody help me?

https://youtu.be/JXc-m6dfHdc

neophyl, 2021-01-17:
First suggestion would be to install the proper full 2.3.0 release rather than a rc and see if the problem still persists.

rtyr, 2021-03-19:
Does it happen with 2.3.0 final?

rtyr, 2021-04-22:
We found out that this bug is related to the new PrusaSlicer version notification. It seems to happen only with Spanish/French language selected. We are working on a fix.

Similar issue https://github.com/prusa3d/PrusaSlicer/issues/6392

lukasmatena, 2024-02-18:
Outdated, closing.

closed 2024-02-18 by lukasmatena