prusa3d/PrusaSlicer #5816: Linux x86_64 Build - prusa-slicer fails to link

TheAndr0id, 2021-01-17:
### Version
2.3.0

### Operating system type + version
Linux x86_64

### Behavior
When building from source the final binary prusa-slicer fails to link with the following message:

```
undefined reference to symbol 'png_create_write_struct@@PNG16_0'
/usr/lib64/../lib64/libpng16.so.16: error adding symbols: DSO missing from command line
```
Yup, I said "right, what the hell does that mean?"

So, after poking Google, I found this page:
https://stackoverflow.com/questions/19901934/libpthread-so-0-error-adding-symbols-dso-missing-from-command-line

Now there's a lot going on, but about half way down, someone mentioned this:

> The DSO missing from command line message will be displayed when the linker does not find the required symbol with it's normal search but the symbol is available in one of the dependencies of a directly specified dynamic library.
> 
> In the past the linker considered symbols in dependencies of specified languages to be available. But that changed in some later version and now the linker enforces a more strict view of what is available. The message thus is intended to help with that transition.
> 
> What to do?
> 
> If you are the maintainer of the software
> 
> You should solve this problem by making sure that all libraries that are needed to satisfy the needed symbols are directly specified on the linker command line. Also keep in mind that order often matters.
> 
> If you are just trying to compile the software
> 
> As a workaround it's possible to switch back to the more permissive view of what symbols are available by using the option -Wl,--copy-dt-needed-entries.
> 
> Common ways to inject this into a build are to export LDFLAGS before running configure or similar like this:
> 
> export LDFLAGS="-Wl,--copy-dt-needed-entries"

So, I added in the -Wl, --copy-dt-needed-entries into the command that linked prusa-slicer and no more DSO missing message.    I think the "keep in mind that order often matters." in the above applies.  I'll see if I can find a way to shoehorn these into the LDFLAGS for the build process since the copy/paste of the whole build line for the final binary was well, not fun.

The binary starts just fine, I haven't did any slicing yet to fully test, but the compile is complete.

bubnikv, 2021-01-18:
> Operating system type + version
> Linux x86_64

What distro? What Linux?

TheAndr0id, 2021-01-18:
It doesn't matter, but it's Slackware 14.2 x64.  gcc is versions 9.2.0, all is x64 as per Slackware standards (Slackware x64 does not support 32bit compatibility by default).

Also, from the ld man page:
```

--copy-dt-needed-entries
       --no-copy-dt-needed-entries
           This option affects the treatment of dynamic libraries referred to
           by DT_NEEDED tags inside ELF dynamic libraries mentioned on the
           command line.  Normally the linker won't add a DT_NEEDED tag to the
           output binary for each library mentioned in a DT_NEEDED tag in an
           input dynamic library.  With --copy-dt-needed-entries specified on
           the command line however any dynamic libraries that follow it will
           have their DT_NEEDED entries added.  The default behaviour can be
           restored with --no-copy-dt-needed-entries.

           This option also has an effect on the resolution of symbols in
           dynamic libraries.  With --copy-dt-needed-entries dynamic libraries
           mentioned on the command line will be recursively searched,
           following their DT_NEEDED tags to other libraries, in order to
           resolve symbols required by the output binary.  With the default
           setting however the searching of dynamic libraries that follow it
           will stop with the dynamic library itself.  No DT_NEEDED links will
           be traversed to resolve symbols.
```
I've never seen or used this option before, but it resolved the linking issues with the final binary.

bubnikv, 2021-09-22:
Are you compiling with static dependencies?
https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md

We do not support non-static builds.

TheAndr0id, 2021-09-23:
That's exactly the guide I followed. Weather you are building dynamically or statically all of the modules that are built still need to be linked into a final binary. It's a static binary, but it's still linked using ld. 

The problem here is there are dependencies between the modules and the static libraries that are at system levels. The "DSO missing from command line" error is due to the libraries not being linked in the correct order. As far as I can tell this is because the static libs are using system calls from other static libs and if they aren't linked in the correct order they cause the linker to throw the above error. 

Please review the link I added in the first message (or view the man page for ld). 

There's really two solutions here. 

1) Determine the correct order for the libraries to pass to the linker. This is the *right* solution, but also the most time consuming and tedious. 
2) Add the "--copy-dt-needed-entries" option to the linker flags when the binary is linked. This works but from what I can see it's a bit of a kluge. I know the man page says "dynamic libraries" but from my understanding it also does the same thing for static ones as well. (see the first link from stackoverflow) 

All I know for sure is that I couldn't get the final binary to link after the build was complete. I added this flag to the LDFLAGS and it works fine. I've used the resulting binary (which is static - I've checked multiple times) for months without any issues. 

[ https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md | https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md ]

bubnikv, 2021-09-23:
All right, I got it.

Would you please try to reorder the libraries in the generated Make file? I suppose the libpng reference needs to be pushed to the left in some of the linking lines. That would help us to identify the culprit and fix it in CMake files.

Thanks.

TheAndr0id, 2021-09-24:
Unfortunately I can't at the moment. My system is in a cardboard box in a storage unit. It may be months before I can run "make" again...

lukasmatena, 2022-04-05:
Is this still an issue?

TheAndr0id, 2022-04-10:
Sorry, I don't know and I have no way to test.  If someone hasn't worked on it, I'd say it's still an issue, but since there's now 2.4.1 (and rcs for 2.4.2) issues with 2.3.0 are a bit of a wasted effort.  When I am able to build 2.4.x, I'll open another ticket if this is still an issue.