prusa3d/PrusaSlicer #5818: Thumbnail/Gcode Preview for Slic3r/PrusaSlicer/SuperSlicer for BigTreeTech TFT Screens

sarvenn, 2021-01-18:
Hi,
Thumbnails generated by Slicer could not be viewed in BTT screens. Codes generated by slicer is not applicable for BTT TFT screens.

In BTT touch screen repository there is a custom solution for that. Please see the link below.
Could this solution be integrated to PrusaSlicer?

#https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1238

bubnikv, 2021-01-18:
Could you notify BTT to update their firmware to be compatible with PrusaSlicer way of storing the thumbnails?

sarvenn, 2021-01-19:
I suppose Cura also uses a plugin to convert the gcode for BTT. Your suggestion also may solve the problem but this time Cura plugin also needs to be updated in my opinnion.
Is there any way to implement and integrate the solution above into PS? Is it possible or applicable?

closed 2021-07-13 by sarvenn