prusa3d/PrusaSlicer #5819: How to modify support logic?

MeerkatSuricata, 2021-01-18:
I am having a lot of issues with Prusa's support logic: it currently sets real contact Z distance as sum of "support_material_contact_distance" and "nozzle diameter" parameters, but I would like it to be "support_material_contact_distance" + "layer_height" or maybe even just the "support_material_contact_distance", theredore my bottom fill ends up 0.4mm away from the support. How can I change it?

To be more specific, you can find a more comperhensive explanation in issue #2930, there is a message by @ruomserg made on 5 Apr 2017

rtyr, 2021-01-18:
>  it currently sets real contact Z distance as sum of "support_material_contact_distance" and "nozzle diameter" parameters

The bottom most surface is printed with bridge flow so it is expected behavior.

> How can I change it?

You can use negative values for the "Z contact distance", for example `-0.1`.

MeerkatSuricata, 2021-01-18:
> The bottom most surface is printed with bridge flow so it is expected behavior.
>
>
> You can use negative values for the "Z contact distance", for example `-0.1`.

Thanks, I’ll try the negative values and report on my results. BTW I’ve seen a lot of people having trouble with this “expected” behavior, as I usually work mostly with S3D, for me this behavior is completely unexpected and unlogical. It shouldn’t be a bridge if it is supported, therefore there should not be the bridge flow setting applied. I think it would be a great idea to add “Disable bridge flow/recognition under supported area”, making the slicer set the Z distance value without adding nozzle diameter to it

MeerkatSuricata, 2021-01-18:
Basically this bridge flow recognition makes detachable supports useless, as the bottom fill just prints in the air without being squished against the support in desirable manner

rtyr, 2021-01-18:
There are already opened issues related to that.

ruomserg, 2021-01-18:
Hi! As an original supporter behind the idea (please note that original issue was for the Slic3r project: https://github.com/slic3r/Slic3r/issues/2930) I'm quite two-minded about this after 3 years of intense 3D-printing.

- Printing with the non-soluble support interface is more like a magic then the technology. Some materials (e.g. PETG) make tight model/support layer bonds. Other materials (e.g. ABS/PLA) make weaker bonds. This also depends heavily on the nozzle temperature, fan speed, overall cooling efficiency and other known and unknown natural factors.
- Even the same kind of plastic could exhibit radically different bonding properties with different color additives and/or accumulated moisture. 
- Modern support generation strategies (sheath around the support in PrusaEdition) almost completely eliminate the perimeter dragging (which was an issue in original Slic3r) still keeping the supports breakable from the model body.

So,

- I still believe that for materials with medium/weak bonding capability it is better to print the first supported layer using normal (non-bridge) mode with an increased layer height and/or reduced flow to control the extrusion contact area (and therefore layer adhesion)
- Also, this could be useful for print materials with large shrinkage coefficients as bridge cooling tend to deform and delaminate the boundary layers.
- With the ability to add/remove supports manually and with a sheath option, most ordinary prints should be successful with ABS/PETG/PLA material with no changes in supports/boundary layers handling (as seen in current Prusa Edition).
- Breakable same-material supports still require tweaking the hot-end temperature and cooling fan speed on spool-to-spool basis.
- If high-quality removable supports are required - the only option is a two-nozzle extruder and a soluble interface on top of the support (also supported by Slic3rPE).

Having said that, I recommend to drop/postpone the issue about boundary layer height calculation and invest time and resources into more useful features. Your opinion may differ.

MeerkatSuricata, 2021-01-18:
> There are already opened issues related to that.

And there is still no sufficient answer. I’ve read them all, and there is always a saying “this is how it’s supposed to be” but for some reason no one can explain why it’s supposed to be that way. S3D support interface layers work just fine, while Prusa’s support always raise questions. So is it really suppose to be that way? Or maybe we should take a closer look at how interface layers really work?

MeerkatSuricata, 2021-01-18:
> You can use negative values for the "Z contact distance", for example `-0.1`.

It kinda worked, but actually it didn't: I managed to get a 0.2mm gap, but for some reason the slicer decided to insert another interface support layer sytraight into bottom fill layer. Here's a pic 
[image]

rtyr, 2021-01-18:
>  I’ve read them all, and there is always a saying “this is how it’s supposed to be” but for some reason no one can explain why it’s supposed to be that way. 

I don't think your interpretation of "this is how it's supposed to be" is completely fair. It is usually about explaing the current implementation of the support generator (which is mainly inherited from the upstream Slic3r) and closing duplicate issues.

I reccommend this issue to read https://github.com/prusa3d/PrusaSlicer/issues/102.

As you probably know, there are several opened issues on this topic, so there is no need to open new duplicates. You can simply comment in some of those issues. For example this https://github.com/prusa3d/PrusaSlicer/issues/5793 issue is 2 days old. I left it opened since the author included some data which we can use for tests.

MeerkatSuricata, 2021-01-18:
> I don't think your interpretation of "this is how it's supposed to be" is completely fair. It is usually about explaing the current implementation of the support generator (which is mainly inherited from the upstream Slic3r) and closing duplicate issues.

There is an explaination how it works, and I get how it works. What I'm trying to tell is that I need it to work in slightly different way, the way every other slicer copes with Z distance. So can anyone tell me how to get it working how I want it to, specifically "I want to set my own EXACT interface Z distance without the slicer adding additional 0.3mm from nozzle diameter"

> As you probably know, there are several opened issues on this topic, so there is no need to open new duplicates. You can simply comment in some of those issues. For example this #5793 issue is 2 days old. I left it opened since the author included some data which we can use for tests.

These issues will be piling on until there is a sufficient and clear answer on how to at least replicate the desired behavior. I am not posting this to annoy you, I'm posting this because there is no clear solution

MeerkatSuricata, 2021-01-18:
> I reccommend this issue to read #102.

Read it and found exactly what I expected: no clear answer, only promises to fix it/look into it.

And a few quotes from @bubnikv:
"I did not invent this, I just inherited it and now I am trying to justify the behavior". Exactly what I said, you're trying to justify this, but not make a frequently asked feature of disabling nozzle diameter addition. 
Another one: "I don' think it is needed or wished. I believe that may of the issues were
due to the bugs / limitations of the previous algorithm." Guess what, it is still wished by many and actually needed for people who mostly print ABS (although PLA useres have same issues too). And it's not a bug, cmon, this issue has been out for three (!) years.

MeerkatSuricata, 2021-01-18:
Currently looking into the source code, found this line. Although I do not know how to code (like at all), my guess that this is the feature I'm looking for.
[image]

ruomserg, 2021-01-18:
This is going to be a complex issue. Users who print with sticky plastics (like PETG) would probably require to keep the current behavior (e.g. fan enabled, bridge type extrusion, extra +0.2mm layer height). PETG tends to be highly adhesive and without bridge mode cooling would make almost unbreakable supports. This way, it needs a bridge/normal mode switch, and then an extra distance value. With a brief glance at the code, I would probably vote for a checkbox to control m_slicing_params.soluble_interface independent of z-contact distance, and then add configured support contact distance ether to bridge layer height or to normal layer height. This would still require adjusting the height and level of the lower interface layer to keep model layers consistent (but it seems to be already done for the soluble interface anyway).

MeerkatSuricata, 2021-01-18:
@ruomserg this is exactly what I would like to see, thank you! A checkbox that says "Turn off bridging support interfacing" or something like that. Is there a way to add this to requested features? I'm not a coder myself, but I would assume it's not gonna take too much effort to implement this.

MeerkatSuricata, 2021-01-20:
> > You can use negative values for the "Z contact distance", for example `-0.1`.
> 
> It kinda worked, but actually it didn't: I managed to get a 0.2mm gap, but for some reason the slicer decided to insert another interface support layer sytraight into bottom fill layer. Here's a pic
> [image]

Guess no one cares. @rtyr can at least this one be flagged as a bug?

rtyr, 2021-01-20:
I don't see any data here, but it looks like the representation of negative z contact distance. You are basically forcing PrusaSlicer to overlap bridge layers with the interface. We allowed negative values based on the demand of the users, who wanted to experiment with contact z distance outside of the standard algorithm (like you). I believe it is not possible to use negative values in the original upstream Slic3r.

This is z contact distance 0.1mm (0.2mm layer height on base model). The thick blue line is the bridge (round extrusion with 0.4mm diameter).
[image]

MeerkatSuricata, 2021-01-22:
> I don't see any data here, but it looks like the representation of negative z contact distance. You are basically forcing PrusaSlicer to overlap bridge layers with the interface. We allowed negative [...]

Yes it is the representation. Yes I realise I'm forcing the overlap, it's intentional. But why twould the slicer insert ONE MORE unnecessary interface layer straight inside the model. I mean with this slice it goes like this from down to top: interface layer > interface layer > gap 0.2 (as 0.3mm nozzle and -0.1 interface Z distance) > bridge layer AND interface layer ptinted on the same height. You can see the support layer overlap the bridge on my screenshot, right? Unfortunately I am not allowed to attach the model, but I saw the interface layer being printed straint into the bridge with my own eyes.

rtyr, 2021-01-22:
If you think there is an issue with the support generation or gcode preview when using negative z values please open separate issue and provide some **3MF project** file where it occurs.

MeerkatSuricata, 2021-01-22:
> If you think there is an issue with the support generation or gcode preview when using negative z values please open separate issue and provide some **3MF project** file where it occurs.

OK. If I get a model I can share I will open a new issue. thanks for the support

closed 2021-01-22 by MeerkatSuricata