prusa3d/PrusaSlicer #5821: Volume non tranché

GCRIBIER, 2021-01-18:
### Version
_Version of PrusaSlicer used goes here_
2.3.0+WIN64
_Use `About->About PrusaSlicer` for release versions_
2.2.0+WIN64
_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_
?
### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
Édition	Windows 10 Professionnel
Version	20H2
Installé le	‎18/‎12/‎2020
Version du système d’exploitation	19042.746
Expérience	Windows Feature Experience Pack 120.2212.551.0

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
PRUSA I3MK3S Avec mmu2s d'origine

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
Il s'agit d'un problème de decoupage , lors du tranchage une partie du modele 3D disparait , il s'agit une zone dessiné avec une distance de 0.2mm permettant de crée du support en manuel

_Is this a new feature request?_
non

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
[DEP 20-169 - bague droite 18° - version elastomère V4.zip](https://github.com/prusa3d/PrusaSlicer/files/5830154/DEP.20-169.-.bague.droite.18.-.version.elastomere.V4.zip)

[Doc1.pdf](https://github.com/prusa3d/PrusaSlicer/files/5830151/Doc1.pdf)

neophyl, 2021-01-18:
Those built in supports are too thin for them to be picked up in the slice. with the settings you have selected.

You could either enable 'Detect thin walls' which makes the feature slice or my preferred method is to change the extrusion width for external perimeters from the 0.45mm default it is currently set to, down to 0.4mm which also results in those features being sliceable.

GCRIBIER, 2021-01-18:
Effectivement , merci pour votre reactivité

closed 2021-01-18 by GCRIBIER