prusa3d/PrusaSlicer #5822: color change in the sidebar

robertocolucci3d, 2021-01-18:
### Version
Version 2.3.0+

### Operating system type + version
  Modellname:	iMac
  Modell-Identifizierung:	iMac17,1
  Prozessortyp:	Quad-Core Intel Core i5
  Prozessorgeschwindigkeit:	3,2 GHz
  Anzahl der Prozessoren:	1
  Gesamtanzahl der Kerne:	4
  L2-Cache (pro Kern):	256 KB
  L3-Cache:	6 MB
  Speicher:	24 GB
[image]
[image]
[Inprinted (10x) NFC Chip.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5830253/Inprinted.10x.NFC.Chip.3mf.zip)

### 3D printer brand / version + firmware version (if known)
PRUSA i3MK3S - Last Firmware

### Behavior
When I put a color change in the sidebar, the marker no longer appears.
In the attached file there is a color change at height 1.80 mm

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Patrickmaartense, 2021-01-20:
I have the same issue under windows.
in the RC1 I could still use that feature
decided to go release today , but will revert back for now as i use that a lot to put in inserts

lukasmatena, 2024-02-18:
Outdated, closing.

closed 2024-02-18 by lukasmatena