prusa3d/PrusaSlicer #5823: Improve Support material adhesion by base type interface layers under soluble supports.

spiky2021, 2021-01-18:
### Version
2.30
### Operating system type + version
Win 10

### 3D printer brand / version + firmware version (if known)
Anycubic I3 Mega, Craftbot Flow IDEX XL
### Behavior
At the moment soluble support material adhesion is weak due to sparse support layers under soluble support layers.
[image]

Now I added two modified methods to the SupportMaterial Class (.cpp and .hpp attached). 
[new_SupportMaterial_Class_files.zip](https://github.com/prusa3d/PrusaSlicer/files/5831412/new_SupportMaterial_Class_files.zip)

[image]
[new_support_files.zip](https://github.com/prusa3d/PrusaSlicer/files/5831221/new_support_files.zip)

This adds two base type interface layers to the support structure, in case the extruders are different and soluble support is choosen. 
Now soluble support adhesion is much stronger.
For someone expired with the GUI implementations, it might be an easy task to add an input for the number of base interface layers.
Though I believe two layers are reasonable for most situations. 

By the way, I believe there is a bug in the original generate_interfaces method. The " idx_top_contact_first" is searched under the condition "layer->print_z >= intermediate_layer.print_z", but it should be "layer->bottom_z >= intermediate_layer.print_z" to avoid excess layers by swapping a layer too later to the upper top contact. In other words, compared to the bottom contact code it should be mirrow symmetric.

_Is this a new feature request?_ yes? 

I am new to the PrusaSlicer framework and GitHub (1 Week experience). Thus I didn't know how to make a pull request and wasn't interested in binding VS2019 to Github. But I hope someone of you can bring it to the base branch.
  

#### Project File (.3MF) where problem occurs

spiky2021, 2021-01-20:
Just a moment ago I saw, that I uploaded the wrong Class files. Some functionallity is commented, sorry for that!
Further I saw that there is still an interface in excess generated. I try to fix this as soon as possible.

spiky2021, 2021-01-21:
Commenting removed and a fix for excess layers done. The index for the top_z/bottom_z discriminators uses the interface layer number. Since indexes usually start at zero, the number needed to be reduced by one.
Original behaviour for two interface layers  requested (actual PS 2.3.0 Version as well):
[image]
And now:
[image]
Supports are generated by painting. One can easily see the two base type support, as well.
New Class files code:
[new_SupportMaterial_Class_files.zip](https://github.com/prusa3d/PrusaSlicer/files/5848050/new_SupportMaterial_Class_files.zip)

A picture of a first print:
[image]
Materials: Extruder Greentec Pro and Polymaker Polydissolve S1.
Stringing and artifacts are generated by undry Polydissolve. Meaning the stiffness against the feeder and the viscosity was wrong. Without base interface layers I couldn't achieve prints with such tiny surfaces like 2x10mm, even with dry filament.

Done with a Craftbot Flow IDEX XL. No wipe tower, just used the wipers of the printer. It reduced the number of head preparation and therefore time and material costs by about 50%, even for this narrow top to bottom distances.   
Supports were generated by painting and XY-Distance to zero, to achieve a support touching to tilted surfaces.
Here maybe soluble interface should be inflated to achieve a touching without this workaround. 
For simple interfaces structures now even soluble support prints with single head printers by adding a pause or material change are possible.

The tiny dot of polydissolve on the bed is introduced to overcome a bug of PS for the multi extruder toolchange.
The generated gcode always introduces a normal enter retract/unretract for the first use of the head, instead of using a retraction procedure defined in a custom toolchange code. It leads to missing or overshoot material in the nozel, that leaves artifacts on the object and even worse deminishes adhesion.

spiky2021, 2021-01-21:
Just for comparison the same part printed with dry filament:
[image]

spiky2021, 2021-01-30:
Now provided as pull request combined with changes under #5903.

bubnikv, 2021-02-15:
Implemented with 32db22b77c73dd1579da686b748267ce5ee1aeb5, based on a pull request by @spiky2021

closed 2021-02-15 by bubnikv