prusa3d/PrusaSlicer #5824: Infamous 3DBenchy hull line triggers layer shift

calini, 2021-01-18:
### Version

2.3.0+

### Operating system type + version

macOS Big Sur 11.1

### 3D printer brand / version + firmware version (if known)

Ender 3 Pro + MicroSwiss Direct Drive + Marlin 2.0.x-bugfix

### Behavior

* Repeatable layer shift exactly at the hull line printing a 3DBenchy with TPU.

* _Settings_: 
0.2mm layer height
low speeds, 30mm/s or lower - tried as low as 20mm
20% cooling - tried values from 0% to 40% 
z hop - tried it both on (1mm) and off 
215C on the hotend, 60C on the bed
1mm retratction @ 20mm/s
Avoid crossing parameters - on

Looking at a gcode viewer, nothing out of the ordinary except a couple of lines going from about the hull line (layers 48-55) to the origin, but watching it live I did not see the printhead move anywhere else.

The machine prints other simple shapes out of TPU flawlessly (like a hollow tube printed pointing up), also prints PLA benchies with no issues (except a very slight hull line, but that's to be expected). No physical issue with the Z axis detected. Multiple other prints tried.
Also trying to slice the benchy with a different slicer (Cura) produces an uglier looking print, as I have not tuned it, but 0 layer shift.

* _Results_: 

This one is the first one I tried, very slight layer shift, the rest of the print looks great
[image]

The rest are all layer shifts at the exact same level
[image]

#### Project File (.3MF) where problem occurs
[3DBenchy.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5831734/3DBenchy.3mf.zip)

rtyr, 2021-01-19:
I don't see any issue in the gcode. Since I was testing some different TPU/TPE related things, I tried to print your gcode (gcode generated from your 3mf project@PS2.3.0).

As you can see, the print was successfull (MK3S+FilaFlex40). I believe it is safe to say, that the output from the PrusaSlicer is OK.

Unfortunately, that's likely all we can do for you. You are in the best position to tune up your print profile or to find the cause of the skipped steps on your printer.

[image]

calini, 2021-01-19:
Hi, thanks! Well at least that excludes the possibility that it's a gcode bug. I had a look through a couple of gcode viewers and have not seen anything particularly weird either. Now my only lead is, the TPU I have behaves weirdly and curls up or expands, catches the nozzle and that causes the layer shift.

closed 2021-01-19 by calini