prusa3d/PrusaSlicer #5825: prusa 2.3.0 crash with 0.25 nozzle

PowerChaos, 2021-01-18:
### Version
PrusaSlicer
Version:   2.3.0-rc3+win64
Build:     PrusaSlicer-2.3.0-rc3+win64-202101081019

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19041), 64-bit edition
Total RAM size [MB]: 17,093MB
OpenGL installation
GL version:   4.3.0 - Build 20.19.15.5144
Vendor:       Intel
Renderer:     Intel(R) HD Graphics 4600
GLSL version: 4.30 - Build 20.19.15.5144

### 3D printer brand / version + firmware version (if known)
Ender 3 V2 + bltouch ( merlin 2.0.1 i guess) 

### Behavior
*Crash when using 0.25 nozzle
**This only apply to the version 2.3.0 as LATEST STABLE , this does not occur in the RC3 version ( where this report is based on ) 

#### Project File (.3MF) where problem occurs
[prusaerror.zip](https://github.com/prusa3d/PrusaSlicer/files/5832212/prusaerror.zip)

*This file also include my printer profile , exported from verion 2.3.0 and imported in the RC3 version
this can be repeated by just using the import config and then the program will crash with that file , STL file also included ( as i use STL instead AMF

Greetings from PowerChaos

rtyr, 2021-01-18:
I can reproduce crash when importing `PrusaSlicer_config_bundle-0.25.ini` using `"Import Config..."` option (`Import Config Bundle...` works).

bubnikv, 2021-01-19:
Fixed with 5294a48ee8bdbf808ed4b9570778657b3c2e15ab

    Importing a config bundle as a config file was broken for
    multi material FDM printers, where a different filament was active
    for each extruder.

Thanks for the report.

closed 2021-01-19 by bubnikv