prusa3d/PrusaSlicer #5826: prusaslicer 2.3 gives error toolpath detected outside print area when working close to machine limit

kjkroeker, 2021-01-18:
### Version

Version 2.3.0+win64

Windows 10 2020

Prusa Mk3S firmware 3_9_0_3421

Imported object fits on the bed and auto aligns without error, but when sliced an  error occurs saying that a toolpath was detected outside the print area. Gcode seems ok but I have not yet printed it. 

[Dust boot  v2.zip](https://github.com/prusa3d/PrusaSlicer/files/5832303/Dust.boot.v2.zip)

[image]
[image]

foreachthing, 2021-01-18:
I disabled the brim and rearranged it. All paths on the plate.
Or, with brim, but 0.5 mm distance to object.

closed 2021-01-18 by kjkroeker

foreachthing, 2021-01-18:
:disappointed: You're welcome.