prusa3d/PrusaSlicer #5827: Synchronized support with object layers even for removable support

eFinger, 2021-01-18:
### Version
2,3

### Operating system type + version
Win10

### 3D printer brand / version + firmware version (if known)
Custom / klipper

### Behavior
I like to use removable support (no second solvable material in use). I like also to have synchronized layers together with the object. But the option "Synchronize with object layers" is only available if I use none offset.
Why? Depending on object and support structure I have quite regular the case that volumetric flow is changes a lot between (because of the big differences on layer hight). Additional the "Cooling" settings are acting negatively. If layer time (e.g. of a small object) is low, the print speed will be reduced. But the "comeback" time to this layer is much longer (because meanwhile e.g. support is printed). So the melt has plenty of time to cooldown already and higher printing speed would be possible.
This leads to some problems and lower print quality. If I use none offset, ok, I could not remove the support (Also not good).
I remember this was not the case in older versions it was possible and I like to have it back. Thanks

 *   Maybe needed adjustment of support height to match well to object shape could be solved within the last layers. But keeping rest of the support synchronized with object.
 *   Another option which will help: introduce a "comeback to layer" time to calculate the cooling behavior (fan/speed). E.g. "comeback to layer" time = Object layer time + support layer time.

eFinger, 2021-01-18:
[image]

Some (simple) example: Nearly every second layer of the support is printed together with object but the other support layer printed only as an own layer. Therefore with layer time for (only) support is very short and speed is breaking down to keep min layer time. This is not the behavior I like (because of too much change in the flow).

rtyr, 2021-01-18:
The support layers are not synchronized with the object layers for two main reasons. It allows the support layers to grow faster than the object layers (if not excesivelly slowed down by the cooling logic) and independent layer height also allows PrusaSlicer to follow the gap between the object and the support top surface precisely. 

You are right about the problem with the cooling logic vs supports, we should improve the algorithm. There is already issue for it https://github.com/prusa3d/PrusaSlicer/issues/984.

The reason to not allow synchronized support layers was that it makes harder to furfill the support to object separation distance. We may consider this to be configurable even for non-soluble supports. There is also issue for that  https://github.com/prusa3d/PrusaSlicer/issues/553.

I am closing this issue, since both problems are already covered by different issues. Your issue will be linked to these issues.

closed 2021-01-18 by rtyr