prusa3d/PrusaSlicer #5828: Feature Request: First Layer: Outline all objects before infill.

DeanDavis, 2021-01-19:
I went search to see if this was available and it's not.
But this paraphrase from a Prusa Forum post crystallizes the need...
"most print failures that take place on the first layer happen during perimeter-printing. And if a print is going to fail, I'd much rather have it happen early on, rather than wait for a boat-load of infill to print, then have it fail on the 10th object while printing the perimeter. Sometimes the first layer alone can take 30+ minutes, so having a failure occur on the last object's first layer perimeters really sucks."
https://forum.prusaprinters.org/forum/prusaslicer/printing-all-first-layer-perimeters-before-starting-on-first-infill/

What made me think of this is I just started a 10h print job with 3 objects. The first layer alone is 1h38m.
I'll have to monitor the job for over an hour before I see if object 3's perimeter is going to be ok.

rtyr, 2021-01-19:
I am afraid that traveling over empty perimeters would create even more problems.

DeanDavis, 2021-01-19:
But doesn't it already? After the print head finishes the outline it has to travel to the start of first layer infill. Is that travel path optimized to not cross the outline already laid down?
If it is, couldn't the first layer path be optimized?
This is just for the first layer so any special optimization would only be in effect for the first layer so impact on total print time would be negligible.

Zimer, 2021-01-24:
I agree this is a needed feature as I just threw away over one hour of print just because of a small object perimeter failure that could save me time and material with this feature.
 
*I arrived here by searching the exact same feature and not sure how I couldn't found anything that was posted earlier to the attached forum post

MasterCATZ, 2021-01-26:
same issue , it seems to work in weird sections , if the perimeters were layed down i would have less chance of filament kicking up when head changes direction if their was something solid their for it to make contact with 

I also wish we could have a round raft option , I can never get it to make a square raft , as it contains no perimeter every layer just kicks up when head changes direction heading back

closed 2026-09-29 by prusaslicer-bot[bot]