prusa3d/PrusaSlicer #5831: Infill Modifier Bug

TheRooster13, 2021-01-19:
### Version
Version 2.3.0+win64

### Behavior
Using a modifier object to modify the infill % causes changes in the bottom layers that have been specified by height-range modifiers.
In this example, the project settings have 0 bottom layers. A height-range modifier adds bottom layers above 1mm. A modifier object is added to the middle of the part, resting on the build plate. The modifier object only changes the infill %, yet when slicing this model, there is a bottom layer on the first layer.
Expected results: No bottom layer on the first layer.

[infill_bug.zip](https://github.com/prusa3d/PrusaSlicer/files/5838308/infill_bug.zip)

This is just a sample file. I know this part doesn't need bottom layers anywhere above the first layer, but the real file I'm working on does need bottom layers higher up. I just don't want bottom layers in the first few layers. The real part should be able to be placed over and object, so it can't have bottom layers.

[image]

_Is this a new feature request?_
Nope

bubnikv, 2021-01-20:
I think the problem here is that the order of application of modifier meshes vs. layer range modifiers is not well specified. You better not overlap the two.

TheRooster13, 2021-01-20:
Neither the modifier mesh nor the layer range modifier specify a bottom layer in the bottom 1mm of the model. So it isn't a simple order issue.
I've also flipped the layer modifier to add a bottom layer in the top half of the model and set bottom layers to 0 for the entire model and the results are the same.
If I disable bottom layers everywhere, then it works correctly. But if there is a bottom layer anywhere in the model (at least in the overall settings or layer range modifiers) the bottom layer will appear.

bubnikv, 2021-06-03:
[image]

It seems to work now in PrusaSlicer 2.4.0-alpha1. The order of application of modifiers is:
1) Layer range modifier
2) Modifier volumes in the order they are listed at the side panel, last wins.

closed 2021-06-03 by bubnikv