prusa3d/PrusaSlicer #5832: Interface layer for supports in broadly planar areas is not adhering correctly.

scottmudge, 2021-01-19:
### Version
2.3.0 

### Operating system type + version
Windows 10 x64

### 3D printer brand / version + firmware version (if known)
Prusa MK3S+

### Behavior
The interface layers between the support and the primary object are not laid down correctly. They are not correctly adhering to the underlying support substrate, and causes very bad quality issues. I'm not sure if it's due to under-extrusion, or due to the print head being too far away from the support layer, or perhaps the start of the interface path does not align with the support layer?

I tried tweaking the support settings -- distance, number of interface layers, etc -- but the same result occurred. I tried printing this object 4 times, with the exact same result:

[image]

#### Project File (.3MF) where problem occurs
[CatCup_3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5838504/CatCup_3mf.zip)

rtyr, 2021-01-20:
I was not able to reproduce the issue. You may try to reduce the speed of the support interface, the interface speed may be too high for the combination of low layer height/extrusion width/filament/temperature. You can also increase extrusion width of the support material ("Advanced" tab).

[image]

Maybe I can slow support interface speed down a little bit in some stock print profiles to make it more reliable with various filaments.

[image]

> Standard PETG printing parameters used.

PETG settings used in your 3MF are not standard, but it is likely not the cause

scottmudge, 2021-01-20:
Thank you for the feedback, i did modify some of the parameters to better suit the PETG material I was using, which had some stringing issues.

But it was probably speed related. Is there a separate speed parameter for the interface layer, or is it a global support material speed? The support itself printed fine, it was only the interface layer that failed.

rtyr, 2021-01-20:
> Is there a separate speed parameter for the interface layer, or is it a global support material speed? 

Yes, you can set separate speed for the interface.

[image]

scottmudge, 2021-01-20:
Thanks, should have checked myself, but was away from my PC. Appreciate you taking the time to look into this, I consider it resolved.

closed 2021-01-20 by scottmudge