prusa3d/PrusaSlicer #5833: Slicing issue

MarcAhoy, 2021-01-19:
2.3.0+Win64

Flat surface in model is being split into 2 separate layers 

[image]
[Test.zip](https://github.com/prusa3d/PrusaSlicer/files/5839063/Test.zip)

Area5142, 2021-01-20:
This is properly due to a rounding error when the model is sliced. If sliced at a layer height of 0.199 or 0.201 the result is:
[image]

MarcAhoy, 2021-01-20:
Thanks, I'll do that from now on

bubnikv, 2021-01-20:
https://github.com/prusa3d/PrusaSlicer/wiki/Frequently-Asked-Questions-(FAQ)#strange-pattern-on-a-flat-surface

closed 2021-03-02 by rtyr