prusa3d/PrusaSlicer #5834: .3mf renders incorrectly, but same part's .stl renders correctly

iovsiann, 2021-01-20:
### Version
_Version of PrusaSlicer used goes here_
2.3.0+Win
_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
PrusaSlicer
Version:   2.3.0+win64
Build:     PrusaSlicer-2.3.0+win64-202101111315

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19041), 64-bit edition
Total RAM size [MB]: 16,556MB
OpenGL installation
GL version:   4.6.0 NVIDIA 451.67
Vendor:       NVIDIA Corporation
Renderer:     GeForce RTX 2060/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
[... 362 lines cut ...]

* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
Import attached .3mf, .stl into Prusa Slicer.

* _Expected Results_
Imported .3mf and .stl look same because both files are same part.

* _Actual Results_
Imported .3mf and .stl look different.

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
See attached.
[image]
[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

[Axle.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5839464/Axle.3mf.zip)
[Axle.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/5839467/Axle.stl.zip)

iovsiann, 2021-01-20:
Update: here is an .3mf file (from Solidworks 2020) that won't even load into Prusa Slicer, but displays correctly in Paint 3D.
[image]
[image]

Area5142, 2021-01-20:
This is properly caused because the model Axle.stl has errors and is not manifold. This can be seen in PS as the warning triangle next to the file in the object tree list.
Analyzing the file in Blender gives the following result:

[image]

iovsiann, 2021-01-20:
@Area5142 thank you for your feedback, but I'd like to emphasize that .stl does renders correctly. It is the .3mf that renders incorrectly (or fails to load altogether). As far as I know, .3mf does not permit storing non-manifold objects.

Area5142, 2021-01-20:
A faulty STL model often renders correctly (as in this case), but has trouble slicing. PS tries to repair the model, sometime with success, but often it fails.
If the failed repaired model is saved into the .3mf file it will properly not display correct when loaded into PS again.
Try repairing the Axle.stl model in a service like NetFabb or FreeCAD before loading into PS and saving to .3mf file. Then the two models should display the same way.

iovsiann, 2021-01-20:
@Area5142 you are right, Prusa Slicer says STL is faulty, non-manifold. Prusa Slicer says so without having to install PS or Blender.

Should I close the issue or leave it open in the sense that when importing a .3mf that contains a non-manifold (even though .3mf cannot contain non-manifolds by design - this is where I'm confused) Prusa Slicer should say .3mf contains an invalid manifold? Or try to fix .3mf like .stl? Or something like that?

[image]

iovsiann, 2021-01-20:
Almost forgot, Prusa Slicer already repairs .stl into a valid .3mf - just import a broken .stl, Prusa Slicer fixes the manifold, then save the project. Prusa Slicer saves its project (with the repaired .stl) as .3mf. So seems like one can do the repairs without using NetFabb or FreeCAD.

Area5142, 2021-01-20:
I'm just a regular user, not employed by Prusa Research.
Yes PrusaSlicer (PS) can often repair a faulty STL model, but the result is not always as expected - like you experienced...
I mentioned Blender, FreeCAD and NetFabb as repair tools as they work better at fixing broken models in a way that is true to the original. Windows also has a build in 3D tool to do repair broken models.
No model should make PS crash, even when it is totally messed up.

closed 2026-09-29 by prusaslicer-bot[bot]