prusa3d/PrusaSlicer #5835: [BUG] One layer is missing while raft_layers>0 and support_material_contact_distance>0

scott0122, 2021-01-20:
### Version
2.3.0

### Operating system type + version
Win10 X64

### Behavior
Box object dimension: 30mm(L)*10mm(W)*1mm(H)

My settings: 
layer height : 0.25
first layer height : 0.25
raft layers : 3

One layer is missing while support_material_contact_distance>0.

https://user-images.githubusercontent.com/39105828/105213811-21771980-5b8a-11eb-9c57-4d7ea93bcebc.mp4

rtyr, 2021-01-21:
The layer is not missing. With Z contact distance > 0, the bottom most layer of the object is printed with bridge flow (0.4mm extrusion in this case). 

[image]

rtyr, 2021-01-21:
There are some issues related to whether this behavior should be changed/improved, for example https://github.com/prusa3d/PrusaSlicer/issues/3425. You can comment there if you are interested in this topic.

bubnikv, 2021-02-24:
Fixed with aee136cedd890ecb5072d9c9215a824fa3f365b1

Raft is newly printed with normal flow and the z-gap between raft and 1st object layer is configurable.

closed 2021-02-24 by bubnikv