prusa3d/PrusaSlicer #5837: Adding Pause M601, View is changed to Color view

koisdigala, 2021-01-21:
### Version
2.3.0 final

### Operating system type + version
MacOS

### 3D printer brand / version + firmware version (if known)
n/a

### Behavior
Adding M601 using the Ruler.
After adding it, it switches to the Color View which is not relevant for Pause.

Would prefer if the View change was NOT triggered when adding M601. I'm guessing this is just inherited from the behavior of M600.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
n/a

rtyr, 2021-01-21:
It is relevant for the visualization and related info.

[image]

koisdigala, 2021-01-21:
Maybe.  I can say that I much prefer to stay in the view I was in to verify that the pause it at the right layer.  I guess you can argue to keep it the way it is now.

The color view has very litte other information though if you have just added a Pause and no color change.

My normal use case is to view the layers, add the Pause just before the bridge layer over the hole (arrow), then continue inspecting the model for other problems (none of which are visible in the Color view).

Ok, so I feel less strongly about this issue, perhaps more strongly about [5721](https://github.com/prusa3d/PrusaSlicer/issues/5721), which prevents you from seeing the time at Pause using the Ruler.

[image]

koisdigala, 2021-02-16:
Ok, I will close this issue just to get it out of the way.  Seems less important.

closed 2021-02-16 by koisdigala