prusa3d/PrusaSlicer #5840: 24 odd circles at layer height 0.8 STL fine on all other slicers

RhepTheSpring, 2021-01-21:
### Version
_Version of PrusaSlicer used goes here_ 2.30 +win64

_Use `About->About PrusaSlicer` for release versions_ product development

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version Windows 10
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known) M3s MMU2s
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_ 24 circles at LAYER 0.8
* _Steps needed to reproduce the problem_ SLICE MODEL SCROLL TO LAYER 0.8
[image]
[Balder Wheel Hub Bobs Design Prt 1.zip](https://github.com/prusa3d/PrusaSlicer/files/5850462/Balder.Wheel.Hub.Bobs.Design.Prt.1.zip)

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2021-01-21:
It is mainly caused by the model. There are quite big gaps between the individual parts. I made quick rework of the bottom part and the issue dissappeared.

**Original model** (with big gaps between individual parts)

[image]
[image]

**Fixed model**

[image]
[image]

RhepTheSpring, 2021-01-21:
Is there a setting within  prusa slicer ? to define the gap needed for a bridge//??
The model was created  in Cad with no gap,  but I can see by your picture it looks like a series of strait lines (splines) which is not how it was created in the cad, but as you know can show in STL files,  and also within Idear Maker and Simplify 3D and just tried curra and  the problem does not exist! how did you do a rework of the model.

rtyr, 2021-03-02:
Maybe you just need to export it with finer resolution. If I remember correctly, I made the main part solid again and then substracted second part from it (boolean difference), so there are no gaps between the two parts.