prusa3d/PrusaSlicer #5841: Support Raft interaction issue

Raynoodles, 2021-01-21:
_PrusaSlicer Version 2.3.0+win64

### Operating system type + version
_OS: WIndows 10 Version 10.0.18363
_PrusaSlicer-2.3.0+win64-202101111315

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
-Describe the problem: I am loading multiple files on a build plate in order to maximize the amount of items printed at once. I found the following issue with support enforcers being larger than the area outlined (#5610 ). I have not printed it, but I figure my results would be similar to #3324.

The new issue I found is how the support rafts interact with the build plate. Specifically:
   - Crossing other support enforcer rafts
   - Crossing the initial layers of other files (not the support enforcers)
[image]

Steps needed to reproduce the problem_
     - I loaded multiple files on the build plate and chose supports For support enforcers only.

Expected Results
     - I expected the slicer to print support enforcers only in areas that were outlined and not have a raft that took so much of the 
        build plate. I also expected the slicer to understand the first layers of files should not have support rafts interacting with it.

Actual Results - See screenshot

Is this a new feature request? No
[Prusa_enclosure.zip](https://github.com/prusa3d/PrusaSlicer/files/5851643/Prusa_enclosure.zip)

bubnikv, 2021-02-24:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3324

closed 2021-02-24 by bubnikv

Raynoodles, 2021-02-24:
@bubnikv How is this a duplicate of #3324 if the problem is showing up in the PrusaSlicer 2.3? #3324 references PrusaSlicer 2.1.1.