prusa3d/PrusaSlicer #5843: Travel moves are not all shown in the Preview and in Prusa Gcode Viewer

LMF5000, 2021-01-22:
### Version
2.3.0

### Operating system type + version
Windows 10 Pro 20H2 (build 19042.746)

### 3D printer brand / version + firmware version (if known)
Creality CR-10 Mini

### Behavior
When the print contains travel moves, they don't always appear in the visual preview. In this case the travel moves are part of my colour changing script (I move the nozzle to the home position, retract the filament manually, insert manually, purge at the home position, then travel back to the print to continue with the new colour).

PrusaSlicer is not rendering the travel moves between the print and the home position. The same bug exists in Prusa Gcode viewer, and also, the two priming lines printed as part of my start script are not visible in either viewer, nor are the travel moves between priming lines and print.

The lines are correctly seen by pronterface and other slicer software.

#### Project File (.3MF) where problem occurs
[Travel view bug.zip](https://github.com/prusa3d/PrusaSlicer/files/5855097/Travel.view.bug.zip)

Here's what Prusa slicer or viewer see:
[image]

[image]

And here are the lines it's missing (visible in pronterface):
[image]

As you can see, it doesn't render the priming lines as print moves, and it doesn't render the two travel lines for the colour change at all.

The reason those lines are important (to people who do manual colour changes) are because they're a god visual indicator for the final check that the layer height of the colour change is correct before commiting to printing.

LMF5000, 2021-01-28:
Has anyone had the chance to review this?

rtyr, 2021-03-02:
I agree that it should be visible in the g-code preview. It is on our todo list.

bubnikv, 2021-07-29:
Fixed in the upcoming PrusaSlicer 2.4.0-alpha1.

[image]

closed 2021-07-29 by bubnikv

LMF5000, 2021-08-03:
Awesome, thanks for fixing it!